Ultimaker/Cura #3647: [Beta 3.3.0] How to update plugins?

probonopd, 2018-04-10:
I read [here](https://github.com/fieldOfView/OctoPrintPlugin/issues/73#issuecomment-380122314) that it should be possible to update plugins, but how? There is no "Update" button...

[image]
[image]

fieldOfView, 2018-04-11:
It does not help that neither the plugin browser nor the plugin manager show version numbers.

LipuFei, 2018-04-11:
Hi @probonopd , there was a bug in plugin upgrade detection and it has been fixed in CURA-5202. Here are the commits:
 - 19937d1be07eb919ae1f6f791216c95fc2ffcdbc
 - 95f4515e93144e5a1c23f9773a2d58b69fcdd0ab

probonopd, 2018-04-11:
Thanks, so looking forward to the next beta. Keep up the great work!

closed 2018-04-11 by LipuFei