Ultimaker/Cura #3649: Cura 3.21: Not able to open any STL file

rivalion, 2018-04-11:
Cura 3.21 Win 10 64bit

I've installed Cura 3.21, it is opening just normal, but when I try to open STL file the app is crushing. It does not matter if I try to open it from the app File menu, or by double clicking. I've tried couple of times reinstalling, but nothing helps.

[cura.log](https://github.com/Ultimaker/Cura/files/1896808/cura.log)

Vandrasc, 2018-04-11:
Hi @rivalion, the crash occurs each time you open ANY stl file? Or reproducible only with one file? If so, could you please share your model with us? Thank you!

rivalion, 2018-04-11:
Hi @Vandrasc I was not able to open any file, so I would say it's not related to a certain model. 
I.e. I've tried to open calibration cube from : www.thingiverse.com/thing:1278865

LipuFei, 2018-04-11:
Hi @rivalion , from the log, it seems to be a problem with file character encoding. Could you tell us which system language do you use on your Windows 10?

fieldOfView, 2018-04-11:
From the log, it looks like qhull is having an issue with the model, before the encoding issue.

rivalion, 2018-04-11:
@LipuFei I have an English version, but with keyboard set to polish layout - "Polski(progarmisty)"

diegopradogesto, 2018-04-25:
I have checked on windows and at least these characters can fail the code above:
`ę` or `ż` (probably some other characters too)

LipuFei, 2018-04-25:
Hi @rivalion , as @diegopradogesto described, I guess you have polish characters in your Windows username? Could you try using a username that only has ASCII characters? At the meantime, we will try to figure out a fix for this.

LipuFei, 2018-04-25:
For dev, JIRA ticket: CURA-5288

LipuFei, 2018-05-02:
Hi @rivalion, I think this problem is now fixed and the fix will be included in the next release. We have created a new testing build with this fix, could you try it out see if it works for you?
https://send.firefox.com/download/48a55ecc2b/#NG5fmL9AuhBzuR_KEsiy4Q

LipuFei, 2018-05-22:
@rivalion I am closing this issue now. If you still encounter the problem, feel free to reopen it.

closed 2018-05-22 by LipuFei