Ultimaker/Cura #3652: Plugin browser send email

fieldOfView, 2018-04-11:
**Application Version**
3.3

Clicking the author name of a plugin opens an email client. As a plugin provider, I would much prefer it didn't, because I do not wish to provide email support. Instead opening a page such as an issue queue or a forum(-topic) would be much preferable.

ianpaschal, 2018-04-11:
A fair point, but that was added because, as I understand it, providing contact info is a defacto requirement for publishing a Cura plugin (as decided by the ancients or whoever).

Paging @Appesteijn since he would know better about whether this requirement is written in stone or if it will likely be phased out in favor of either a different contact solution defined by UM or by whatever the plugin creator would like.

Appesteijn, 2018-04-11:
The main goal was that users know who to contact if they have support issues or questions. Sending them to a forum is also fine, but I think at this moment the link is hardcoded with mailto://

fieldOfView, 2018-05-02:
I am now getting multiple emails from people wanting email support. I don't want to do email support, and as far as I am aware I never agreed to providing email support for the plugins I write. Please fix this.

I have no problems providing support for my plugins, I just don't want to use email to do so.

ChrisTerBeke, 2018-05-02:
Made a PR on our internal packages repo to remove your email address.

fieldOfView, 2018-05-02:
Whilst you are doing that, could you also change my name to fieldOfView in the new system?

ianpaschal, 2018-05-04:
Sorry about that @fieldOfView. I put your name since it's "more correct" and publicly visible on your GitHub profile. I've just merged the PR request making these changes.

ianpaschal, 2018-05-04:
@fieldOfView, by default if there is no `author.email` property, the "Contact" link becomes a "Website" link to the `package.website` which in this case is the GitHub page of your plugin. Is this OK? I am a bit biased against forum links because they have a nasty tendency to get destroyed every time some web admin decides to upgrade the forum software *cough* Unity3D *cough cough*

fieldOfView, 2018-05-04:
Thanks for the changes.

Yes, the github page of my plugin is what I would expect. Note that I am fully aware that my name and email address are publicly available, I just don't need them to be the default options.

ianpaschal, 2018-05-04:
👍 Then you'll see the changes in a few minutes once the API's cache is repopulated from the repo.

fieldOfView, 2018-11-25:
Marketplace links to my email address again. Shall I open a new issue or can someone reopen this one?

ianpaschal, 2018-11-25:
I will reopen it and let Cura team decide how to handle it. I haven't looked at it but first guess is that the old logic used author email if available, and plugin website if not. In the old system we removed your email in your author.json file so that it would use the plugin site link. With the new developer profile author email is never missing, so that's being used now.

nallath, 2018-11-26:
I've put it on our "to discuss" list. I expect that we will fix this next release (I also got emails and I also don't plan to provide email support)

fieldOfView, 2018-11-26:
Thanks. Also good to know I am not the only one.

Appesteijn, 2018-11-27:
https://github.com/Ultimaker/Cura/pull/4891

fieldOfView, 2020-05-08:
Marketplace links to my email address again, from the "installed" tab. Shall I open a new issue or can someone remove the "fixed" label?

mahtDFR, 2020-05-08:
Was this ever fixed? I just tried it in a few versions and the behaviour seems the same. I'll put it up for discussion.

mahtDFR, 2020-05-11:
@fieldOfView completely understand the need for this, however our PO wants to be sure people are going to provide a contact method that is monitored and does not go down a black hole.

So as I understand it we need to modify this snippet of qml:

`            Label
            {
                text:
                {
                    if (model.author_email)
                    {
                        return "<a href=\"mailto:" + model.author_email + "?Subject=Cura: " + model.name + "\">" + model.author_name + "</a>"
                    }
                    else
                    {
                        return model.author_name
                    }
                }`

And remove the hardcoded mailto. But we also need to think of a way to get contributors to choose their preferred contact method in the portal itself.

Ghostkeeper, 2020-05-12:
That PR only removed 1 out of the 3 places where the e-mail was shown or linked:
* The detail page of a package, now removed.
* The author page, which is only visible for materials.
* The installed tab.

The other two were both added before the first one was removed. So no, it doesn't seem that this was ever fixed.

Is it required to have a website? We could show that instead, if it is, but I'm not sure if that's required to submit.

mahtDFR, 2020-05-12:
There's an input for a website for each submitted package, but there's also a website and email address associated with the developer account. I guess the UX guys should also be informed about this since it spans over to the contributor portal as well.

[image]

fieldOfView, 2020-05-12:
Wouldn't it be most logical if these seemingly "fieldOfView" links at least open the same url?
[image]

fieldOfView, 2022-09-09:
I can no longer find mailto: links in the new marketplace, so this issue can finally be laid to rest!

closed 2022-09-09 by fieldOfView