Ultimaker/Cura #3654: problem covered holes

carleas, 2018-04-11:
HI!

I have a problem with cura 3.2.1 i import a model with holes and the program cover this.
I import diferents models, with everyone do the same.
The model doesn't have issues if i check  with X ray.
If I fixing the model in the floor the problem persist.

I hope an answer as soon as possible.

Tanks

PD: attach diferents pictures with the problem 

[image]

[image]
[image]

smartavionics, 2018-04-11:
Hi, it may be a problem with the model, can you please make that available so we can check it. Thanks.

BagelOrb, 2018-04-12:
Try fiddling with the mesh fix settings. Try disabling Union Overlapping Volumes.

carleas, 2018-04-15:
How can i make model available?
Sorry but i have the program with another lenguage and i don't find where stay this option?

Tanks

smartavionics, 2018-04-15:
Zip the STL file and then attach the zip file to this issue.

BagelOrb, 2018-04-15:
Better yet, save the .curaproject.3mf file, rename it to zip and attach
that.

carleas, 2018-04-17:
[Hole problem.zip](https://github.com/Ultimaker/Cura/files/1919279/Hole.problem.zip)
[Hole problem.zip](https://github.com/Ultimaker/Cura/files/1919280/Hole.problem.zip)

smartavionics, 2018-04-17:
If you turn off "Union overlapping volumes" then the hole appears.

Freecad reports this model as having self-intersections, perhaps that is the reason that Cura has a problem with it.

carleas, 2018-04-17:
Solver tanks!

I turn off Union overlapping volumes and i import te model in format .obj now all correct!

Tanks!

closed 2018-04-19 by BagelOrb