Ultimaker/Cura #3656: [3.3.0 beta] All temperatures set to 0 midprint

kirilledelman, 2018-04-11:
Ok, this is a really bizarre one.

I have a dual extruder Mendel type printer. Using "Custom FDM Printer" with hand-calibrated settings but no custom gcode.

Any print involving both extruders will always set all temps - E0, E1, bed to 0 in the middle of the print for no apparent reason. 

* Despite this, print continues, but eventually either the print will come off the plate, or plastic will stop coming out.
* It always happens about 12 minutes from the start.
* Saving out generated gcode and printing using a Pronterface prints fine without temp problems.
* I've disabled screensaver/screen off feature of my Mac - no effect.

smartavionics, 2018-04-12:
Hi, could you please make the gcode available so we can take a look? Thanks.

fieldOfView, 2018-04-12:
@kirilledelman, are you using the new extruder preheating functionality? There's a bug in that, which is fixed here: https://github.com/Ultimaker/Cura/pull/3620

smartavionics, 2018-04-12:
Hi Aldo, I had forgotten about that, it sounds like the culprit.

kirilledelman, 2018-04-12:
@smartavionics @fieldOfView 
Thank you.

Since I can't build the app myself from latest source, would canceling all preheating after it reaches the goal temps and then starting the print prevent this from occurring?

kirilledelman, 2018-04-12:
I tried that, and it just sits there forever. "Printing ... 0%", all the temps are at goals, nothing is happening. BTW the goal temps displayed in Cura are all 0, and printer itself shows as 200/200/72.

Appesteijn, 2018-04-17:
CURA-5227

LipuFei, 2018-04-19:
The pull request has been merged into 3.3

closed 2018-04-19 by LipuFei