Ultimaker/Cura #3661: 3.3.0 beta - Needs a prime after the initial retraction

amagr0, 2018-04-12:
**Application Version** - CURA 3.3.0beta
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Add a custom 3D printer, generate a gcode, and print.

**Actual Results** - In the beginning of the print there's a lack of filament extrusion because of the initial retraction (before the start of the print).

**Expected results** - Prime after the initial retraction.

[image]

There's other similiar situation, but with a dual extruder 3D printer.
If the print start's with the second extruder, it will do the initial retraction and also the "Nozzle Switch Retraction Distance".

[image]

smartavionics, 2018-04-12:
Hi, in your first example above, there is a prime because the extruder is going from -1.5 to 0 on line 28. Sorry, I have no experience of multi-extruder so I can't comment on the second example.

amagr0, 2018-04-12:
That's correct.
This situation only happens for dual-extruders, the problem is in the "G92 E0" before the initial of the print.

BagelOrb, 2018-04-19:
This was the reason we introduced the new gcode for priming an extruder the first time: G240 (?). The problem is inherently unsolvable using normal gcode because the gcode doesn't know where the filament is or where it should be.

I propose you or somebody from the company which produced your printer makes the code changes needed and contribute to Cura if needed.

Ghostkeeper, 2018-05-08:
It's not priming in your second example. Cura assumes that the filament starts at the tip of the nozzle (or that `T1` makes it so). If it doesn't, I suggest that you use a skirt.

closed 2018-05-08 by Ghostkeeper

amagr0, 2018-05-17:
But if there's a retraction before the T1 starts to extrude, how can Cura assume that filament is on the tip of the nozzle?

Can always be solved with the Skirt, but the filament extruded in the Skirt could not be enough since the Switch Nozzle Retraction is 10 mm of filament (on Ultimaker for example is even bigger, 16 mm).

BagelOrb, 2018-05-22:
But there is no retraction before it is switched to T1.

The retraction happens after we switch to T1, so the gcode assumes that the
filament of T1 is configured such that E0 coincides with the filament being
at the nozzle.

kihope, 2018-09-10:
i have dual extruder and still same problem on 3.4

Maxwellfire, 2018-11-26:
In my case at least, I'm trying to do a print without T0 entirely. T0 is disabled in the UI, and should never be touched by the slicer. This is confirmed by the first line of generated gcode being T1. At no point is T0 ever selected. The filament in my printer starts at the tip of the nozzle, cura then retracts it by the nozzle switch amount, and then immediately attempts to print with that same nozzle.