Ultimaker/Cura #3662: Master branch (3.3) - still a problem with gcode file name not being updated.

smartavionics, 2018-04-12:
**Application Version**

Master branch

**Platform**

Linux

**Qt**

**PyQt**

**Display Driver**

**Printer**

**Steps to Reproduce**

Load a model, slice it, clear that, load another model, slice.

**Actual Results**

Quite often the file name shown in bottom rh corner doesn't get updated when the second model is loaded. If I then clear the buildplate again and reload the second model, the file name gets updated.

**Expected results**

When I clear the build plate and load a new model, the file name should be changed to reflect the new model name.

**Additional Information**

diegopradogesto, 2018-04-13:
I can't reproduce it in Windows. Let's see if somebody else in the team can reproduce it in Linux.

closed 2018-04-15 by smartavionics

Ghostkeeper, 2018-05-04:
Also cannot reproduce it on Ubuntu 18.04. I don't think this is OS-related though.

diegopradogesto, 2018-05-04:
I think this or a similar issue was fixed a couple of weeks ago in CURA-5246.