Ultimaker/Cura #3665: [3.3.0B] Settings are difficult to edit, drops values without warning, and is very slow

zarthcode, 2018-04-12:
**Application Version**
3.3.0-beta

**Platform**
Windows

**Display Driver**
Nvidia 391.01

**Steps to Reproduce**
1. Select an input dialog, (e.g. Shell->Horizontal Expansion) and enter a value, then quickly click another value or press tab to move to the next box.
or
2. Open the post processing plugin. Select "Pause at height".  Select Layer No, then attempt to change the layer. 

**Actual Results**
1. Input SILENTLY disappears/reverts to previous value.
2.  Pressing enter closes the dialog, tab does nothing, clicking on another input value reverts.

Usually, waiting on a slice to start before attempting to move the cursor elsewhere actually stores the value.

Sometimes, if I'm not done typing, a slice will start (or, that's what I suspect), and my current value will be taken as input - even though I haven't pressed tab/enter - focus is temporarily lost during the slice

**Expected results**
I press tab, move to the next box, and happily/speedily input values w/o worry of lost data.  If I want to revert, I'll like the revert button. If the application is doing it for me, interrupt me w/a dialog.

and don't re-slice until I've pressed TAB/ENTER/ESC, or clicked elsewhere.

smartavionics, 2018-04-12:
I believe all this badness has been fixed for the 3.3 release.

DimiBaudonck, 2018-04-14:
I have the same issues. Values regularly not being accepted and just revert back to the previous value. Seems to be random and without warning.

ianpaschal, 2018-04-16:
Correct @smartavionics, this has been fixed and will be in release 3.3.0 (non beta).

closed 2018-04-16 by ianpaschal