Ultimaker/Cura #3668: Cura 3.3.0 BETA - OctoPrint Plugin error while pressing "Print with Octoprint"

raidolo, 2018-04-14:
As topic says.... I can't send the print to my Octoprint server with Cura 3.3 BETA on my MAC 

Plugin Version: 3.3.1 

```
Raids-Mac-Pro:OctoPrintPlugin raid$ pwd
/Users/raid/Library/Application Support/Cura/3.3/plugins/OctoPrintPlugin/OctoPrintPlugin
Raids-Mac-Pro:OctoPrintPlugin raid$ grep version plugin.json
    "version": "3.3.1",
Raids-Mac-Pro:OctoPrintPlugin raid$
```

@fieldOfView pretty simple, here is the error I got on Cura 3.3.0 BETA and its Octoprint Plugin:

```
2018-04-14 22:23:08,836 - ERROR - [MainThread] UM.Logger.logException [81]: Exception: Unable to write to file PI3_RCMicroNut: 'BuildPlateModel' object has no attribute 'activeBuildPlate'
2018-04-14 22:23:08,839 - ERROR - [MainThread] UM.Logger.logException [85]: Traceback (most recent call last):
2018-04-14 22:23:08,842 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/Qt/Bindings/OutputDeviceManagerProxy.py", line 122, in _writeToDevice
2018-04-14 22:23:08,845 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/Users/raid/Library/Application Support/cura/3.3/plugins/OctoPrintPlugin/OctoPrintPlugin/OctoPrintOutputDevice.py", line 302, in requestWrite
2018-04-14 22:23:08,847 - ERROR - [MainThread] UM.Logger.logException [85]:     active_build_plate = Application.getInstance().getBuildPlateModel().activeBuildPlate
2018-04-14 22:23:08,850 - ERROR - [MainThread] UM.Logger.logException [85]: AttributeError: 'BuildPlateModel' object has no attribute 'activeBuildPlate'
```

I also tried to copy over the cura 3.2 Octoprint Plugin files to the 3.3 plugin folder, but it does not work :)
 
Thanks a lot in advance for any fix.

Cheers, 
Alessio.

fieldOfView, 2018-04-14:
Thanks for reporting. There’s a fixed version available, but upgrading plugins is broken in the beta. Please uninstall and reinstall the plugin.

raidolo, 2018-04-15:
I’ve already done it.

Did you fixed it after my post?

If so, I’ll report back if it has fixed the issue and thanks a lot for the fast fix ;)

fieldOfView, 2018-04-15:
Something weird happened when I packaged the 3.3.1 version of the plugin. I'll repackage it tomorrow.

fieldOfView, 2018-04-15:
Can you test http://files.fieldofview.com/temp/v4/OctoPrintPlugin.curaplugin ?
Download the file and drop it into Cura. Then restart Cura.

raidolo, 2018-04-15:
It works. Thanks a lot!

diegopradogesto, 2018-04-16:
Thanks guys.

closed 2018-04-16 by diegopradogesto