Ultimaker/Cura #3669: Middle button click add support blocker

mgth, 2018-04-14:
**Application Version**
3.3 Beta

**Platform**
Windows

**Display Driver**
Radeon 18.3.3

**Printer**
Tevo TORNADO
[CFDMP_Piece2^Assemblage1.zip](https://github.com/Ultimaker/Cura/files/1911040/CFDMP_Piece2.Assemblage1.zip)

**Steps to Reproduce**
In support blockers mode try to move the view with middle button

**Actual Results**
Sometimes it moves, sometimes it adds a blocker

**Expected results**
should just move the view

**Additional Information**

fieldOfView, 2018-04-15:
Nice catch. I don't have a middle mouse button.

Fixed in https://github.com/Ultimaker/Cura/pull/3545

ianpaschal, 2018-04-16:
Thanks @fieldOfView! I think it was expected that I would review your PR since I reviewed the last support blocker PRs but as you know I've been writing a lot of code myself for the last 2 weeks. Someone will get to it ASAP though.

closed 2018-05-04 by Ghostkeeper