Ultimaker/Cura #3670: FR - Make multi color part easy

mtmello2003, 2018-04-15:
Hello,

There are a lot of multi colors print over there (using single or multi nozzles) but the only feature to make multi color in cura is the MERGE function that demands the part be divided out from cura.

Add new features to make multi color to a part that is over the cura bed could be great! For example: option to change extruder by layer (yes, I can use post processing script to do that but in this case, start/end extruder gcode wont be added...) or maybe set "cube" areas and select extruder to this "cube" in the part.

Just ideas to think about. The main point is make a part multi color using only the slicer, not 3d softwares .

Tks

fieldOfView, 2018-04-15:
See the mesh type in Per Model Settings, this (merged) PR has an animation which shows a little bit how to use it:
https://github.com/Ultimaker/Cura/pull/2758

You basically load another model (can be a cube or anything), and set its Mesh Type to "Modify settings for overlap with other models". Then you can set the extruder of this secondary mesh, and overlap it with your normal.

mtmello2003, 2018-04-15:
The Per Model  Settings can be a great solution to use with a multi parts but when I have just a single part to be printed, I can´t see this working to make this single part in a "multi color part", using only the cura.

That is my point: There isn´t features in a easiest way to make a part (mono color) into a multi color.

fieldOfView, 2018-04-15:
Please read the comment again, and look at the whole gif in the linked pr.

The secondary mesh is used to mark areas of the primary mesh to be printed with another extruder (or with different infill, or other settings). This is *way* more versatile than being able to change at certain layer heights.

ianpaschal, 2018-04-16:
@fieldOfView is right. The existing solution is already far better than changing extruder at a certain layer height, and the alternative, which in my mind would mean some sort of way to edit the model into different colored parts in Cura is not even remotely feasible.

Sorry, @mtmello2003, this is a "won't do."

closed 2018-04-16 by ianpaschal

mtmello2003, 2018-04-16:
First of all, I need to thank you all for the time to answer my questions and requests!

About the subject of this thread, IMHO dual or even mult color printers are growing up slower than the monocolor printers because the softwares at all do not give us easy and fast experience of use like mono does.

To print a color part demands - tech - efforts by user and sometimes I left just because of it... We need to slice/separate the part in another modeling software and it is not a piece of cake for someone (like me fo example).

Maybe in the (near I hope) future, the slicers softwares can make this job easy in anyway but now, thats my first impressions using multi color printer...

ianpaschal, 2018-04-16:
> We need to slice/separate the part in another modeling software and it is not a piece of cake for someone (like me fo example).

We could also make such functionality but it would:
- Be nearly impossible to implement without destroying the integrity of the mesh
- Not work nearly as well as doing it with the original CAD file
- Still be very complicated for the end user to do correctly

slivah, 2019-03-17:
Can a kind of brush, with material option and size of the brush, would be possible and easier to program... lets says that instead of modifying the stl model, it would just tell the slicer to use "materialX" at the specific region and "materialY" for the other region.

Ghostkeeper, 2019-03-18:
That's a bit of a difficult problem though. Because a brush can paint on surfaces, not in volumes, and you need a certain volume to be printed with the other material. How deep does the brush go? And how can you paint any deeper than that if you want to?

We're looking into a solution similar to the interface of IceSL combined with a research project that Ultimaker did last year to paint volumes (initially for infill density), but it's going to take a while to come to fruition.

slivah, 2019-03-18:
Well in cura it is possible to scroll layer by layer to see the printing process, the brush could work the same way, let make a 3d brush that can be configure layer by layer... lets says your want 10 layer height of "material#2", you choose the brush size that you want with the "material#2" as the material to be used by the brush and the layer depth that you need (1 to maximal). Lets say that "user" need a 45° angle in the brush, well the user would have to rotate the object 45° to brush and then the "user" will rotate the object back to normal with the brush already impregnated in the object.

It as to be simple at the beginning, but as a "user" i would really enjoy to be able to simply modify object with multi-filament requirement without having to learn 10 more software.

Ghostkeeper, 2019-03-19:
We're of course careful to not go too deep into the CAM software territory. That's a slippery slope.

Scrolling through the layers requires slicing first. That's something we'd like to prevent. Near-plane rendering is more towards the options we've been looking at.

crabbedh, 2023-02-19:
I'd like to add my 2cents as a College instructor and a user of PSlicer, just coming to Cura (and S3 dual ex) to do a simple color change based on layer for highlighting last 2 layers for lettering on model. Setting a support blocker and then secondary mesh to exactly the height in mm is just so much more process to learn and discover than simply setting layers 1-200 to extruder 1 and 201-202 extruder 2.. (then maybe 203+ back to extruder 1). I read the comment that this is "not even remotely feasible".. but you are letting the technology drive your features. It took hours to figure this out first.. and then going back into the fusion 360 model to determine exact Z heights to set ..   And maybe this is  just a normal learning curve, but this technology should be as accessible as possible. Even though I realize some things just have a certain minimum "bar" of understanding as their requirement for entry. 

I will say that Prusa Slicer's layer bar on the right, showing exact height per layer would have been helpful in me setting the exact Z height for my modifiers.. I'm sure it is easy to find somehow in Cura, but I didn't as a new Cura user. The fact that I can hover over the layer slider in PSlicer, and use my scroll wheel to quickly find the layer I want the color change and then click a button is amazing, compared to the process in Cura. (from simply a user perspective) And there can be a zillion technical reasons why this can't work in Cura.. but it is dead simple in PS. If I want inset lettering to be a different color (I want 2 layers that are buried 5 layers from top to be a different color), in Cura, I've got to find the exact Z's, and create 3 sets of modifier blocks, resize them, position them, etc. Then, what if I import 10 more such models that I want exactly same Z modifications? Is there an easy way to duplicate that.. maybe so .. but again.. not easy to find.. (I don't know if an easy solution exists).. 

So I'd really rethink the change color at layer height discussion (with single and dual extruders). I'm sure many are very tired of it as it has come up quite often. But that should tell you that is the UI that can make people more productive with their machines.  I could not believe, coming from PS, that I'd have to go through the process I did to simply set the top 2 layers to use extruder 2. 

So I realize that you don't want to scroll thru layers as a philosophical 'tact'.  But once you do, there are a lot of advantages you can bring to your UI that make some things dead easy for the user/owner of a 3d printer.