Ultimaker/Cura #3671: Load/Unload Filament

tenten8401, 2018-04-15:
**Application Version**
3.2.1

**Printer**
Printrbot Simple Metal

**Expected results**
I think we should be able to have a way to load/unload filament for printers without an LCD controller like my Printrbot Simple Metal, whether that be straight up manual extruder control or user-friendly load/unload buttons.

**Additional Information**
Here's a quick mockup of what I was thinking:
[image]

fieldOfView, 2018-04-15:
This is on my todo list for Cura 3.4. Probably somewhere between proposal 1 & 2

jackha, 2018-07-03:
It's not on the todo list of the Cura team.