Ultimaker/Cura #3677: preset display flow inconsistency (bug?)

filipgoc, 2018-04-16:
**Application Version**

3.2.1

**Platform**

Mac

**Qt**

**PyQt**

**Display Driver**

**Printer**

**Steps to Reproduce**

In my profiles where I have never overridden flow,
- the UI shows flow as 100%
- when sliced, the summary at the end of gcode does not show any material_flow override (which would correspond with 100%)
- BUT the preset summary in the UI shows 90% flow. 

Looks like some sort of display dependency bug? 

**Figures**

[image]

[image]

```
;End of Gcode
;SETTING_3 {"extruder_quality": ["[general]\\nversion = 2\\nname = spring-01\\nd
;SETTING_3 efinition = imade3d_jellybox\\n\\n[metadata]\\ntype = quality_changes
;SETTING_3 \\nquality_type = normal\\nextruder = fdmextruder\\nsetting_version =
;SETTING_3  4\\n\\n[values]\\ncoasting_enable = False\\ncool_fan_full_at_height 
;SETTING_3 = 1\\ncool_fan_speed_max = 80\\ncool_fan_speed_min = 45\\ngradual_inf
;SETTING_3 ill_step_height = 1\\ngradual_infill_steps = 1\\ninfill_pattern = cub
;SETTING_3 ic\\ninfill_sparse_density = 65\\nmaterial_print_temperature = =defau
;SETTING_3 lt_material_print_temperature\\nmaterial_print_temperature_layer_0 = 
;SETTING_3 =material_print_temperature\\nretract_at_layer_change = True\\nretrac
;SETTING_3 tion_amount = 2\\nretraction_hop = 0.15\\nretraction_hop_enabled = Tr
;SETTING_3 ue\\nroofing_layer_count = 1\\nskin_line_width = 0.48\\nskirt_brim_mi
;SETTING_3 nimal_length = 300\\nsupport_angle = 80\\nsupport_bottom_height = 0\\
;SETTING_3 nsupport_connect_zigzags = False\\nsupport_infill_rate = 20\\nsupport
;SETTING_3 _interface_density = 50\\nsupport_interface_enable = True\\nsupport_i
;SETTING_3 nterface_height = 2.5\\nsupport_interface_pattern = lines\\nsupport_i
;SETTING_3 nterface_skip_height = 0.1\\nsupport_use_towers = False\\nsupport_xy_
;SETTING_3 distance_overhang = 1\\nsupport_z_distance = 0.2\\nz_seam_relative = 
;SETTING_3 True\\nz_seam_type = sharpest_corner\\nz_seam_x = 5\\nz_seam_y = 5\\n
;SETTING_3 \\n"], "global_quality": "[general]\\nversion = 2\\nname = spring-01\
;SETTING_3 \ndefinition = imade3d_jellybox\\n\\n[metadata]\\ntype = quality_chan
;SETTING_3 ges\\nquality_type = fast\\nsetting_version = 4\\n\\n[values]\\nlayer
;SETTING_3 _height = 0.2\\nlayer_height_0 = 0.3\\nmaterial_bed_temperature = 60\
;SETTING_3 \nmaterial_bed_temperature_layer_0 = 65\\nretraction_combing = noskin
;SETTING_3 \\nsupport_enable = True\\nsupport_type = everywhere\\n\\n"}
```

BagelOrb, 2018-04-25:
Does this still happen in Cura 3.3 (just released)?

Ghostkeeper, 2018-05-11:
The profile you have selected (IMADE3D JellyBOX / Fine / 0.4mm nozzle) sets the flow to 90%. If I load up that profile it shows this also in the interface, 90% flow. And in the profile manager too. If I create a custom profile for it then it still shows 90% flow everywhere.

Are you sure that it displays 100% flow in the settings list in the side bar? If it does, that is a bug, but I can't reproduce that.

filipgoc, 2018-05-11:
I missed the 3.3 note. I'll explore this within a few days again. Thanks everyone!

ChrisTerBeke, 2018-05-23:
Any luck in 3.3?

filipgoc, 2018-05-24:
@ChrisTerBeke It's gone in Cura 3.3. So whatever it was... let's close. Thanks for following up!

closed 2018-05-24 by filipgoc