Ultimaker/Cura #3678: cura won't slice

seralkis, 2018-04-16:
hello, I need help cura won't slice even when switched off antivirus and firewall
[stderr.log](https://github.com/Ultimaker/Cura/files/1916697/stderr.log)

seralkis, 2018-04-17:
[curaproject.zip](https://github.com/Ultimaker/Cura/files/1920827/curaproject.zip)

diegopradogesto, 2018-04-18:
It slices fine in Cura 3.2 and Cura 3.3. I didn't check any other versions. 
[image]
Maybe try to start with a clean install by removing the configuration folder located in:
- C:\Users\\<your username>\AppData\Roaming\cura\<Cura version> (Windows)
- $USER/Library/Application Support/cura/<Cura version> (OSX)
- $USER/.local/share/cura/<Cura version> (Ubuntu/Linux)
NOTE: Removing this will delete your printers and custom profiles. You can backup your data first.

seralkis, 2018-04-18:
unfortunately it will not help, I've already done it (repeatedly) before writing on this forum

ianpaschal, 2018-04-23:
Please fill out the issue template, especially including, based on Diego's experiments, your Cura version. Might be that we already fixed the issue if you are running a version older than 3.2.

ianpaschal, 2018-04-25:
Closing because author has not filled out the template with info required to debug.

closed 2018-04-25 by ianpaschal

Ghostkeeper, 2018-05-11:
Reading the log indicates that the user is getting an encoding error when CuraEngine outputs a weird character byte to the log (`\xC0`). This indicates that we may be outputting CuraEngine's log in a different encoding than how it's interpreted on the front-end's side. Only we currently don't specify `-fexec_charset` so it should default to UTF-8, which is how it's interpreted in the front-end. So that seems to be working fine.

For now I've made the front-end more robust against these faulty encodings from the back-end: https://github.com/Ultimaker/Uranium/commit/340c586dd67328e1560a261a96fdbf6acfc27866

It should not break on this anyway.