Ultimaker/Cura #3679: GUI is off the screen

JonHeron, 2018-04-16:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

N0YHR, 2018-04-16:
If you don't know to hit Control-E to get the Print interface, there are no clues on the interface.

It's overly complicated, and worse yet, the bed heats, and then just sits there.

I have a highly modified Rigidbot Regular with all the Peter Stoneham mods and prox sensor for bed leveling.

fieldOfView, 2018-04-17:
@JonHeron: we need more information. What resolution are you running Cura on? A screenshot could be helpful.

@N0YHR: Cura only loads with the sidebar hidden if you have hidden it yourself before closing Cura.
If you have more problems, please open other issues, but don't hijack an existing issue.

N0YHR, 2018-04-17:
Jon Heron: Very true regarding hiding the interface.  Once hidden though there is no clue to its existence without finding it in Help.  My screen resolution is 1280x1024.

 

My apologies for hijacking the issue…

diegopradogesto, 2018-04-17:
As @fieldOfView says, please send a screenshot to know what is going on with the GUI

ianpaschal, 2018-04-17:
@N0YHR: This is a separate issue, I believe. We could maybe make a menu item or some other affordance to unhide it, but having the sidebar be hidden is intended behavior.

As for the original issue, it sounds like @JonHeron's screen/window is too small and it's clipping off the bottom of the UI.

JonHeron, 2018-04-17:
Resolution is 1920 x 1080 (16x9)
The machine is an Asus UX430U ultrabook running F27_64
<img>http://theherons.ca/album/var/albums/3D_Printing/Screenshot%20from%202018-04-17%2008-49-06.png?m=1523969508</img>
Cheers,
Jon

JonHeron, 2018-04-17:
Is there a way to set the size of the gui at runtime by chance, from the terminal for example? 
Is Qt used for the GUI?
The font is quite large as well, If I reduce the font size it will likely all fit on the screen too but I cant find  a way to change the fonts size either...
Cheers,
Jon

ianpaschal, 2018-04-17:
I'm just eyeballing this but it looks like Cura is failing to recognize your screen DPI correctly and interpreting the screen as 520 px tall, clipping off 80 or so px of the min window size. Will look into it more tomorrow.

JonHeron, 2018-04-17:
Thank you!
Jon

fieldOfView, 2018-04-18:
Qt seems to think you have a "high dpi" screen. Try setting the env variable QT_SCALE_FACTOR to 1 (or other values) before launching Cura.

JonHeron, 2018-04-18:
Ithink this is a "high DPI" screen but I am not sure I know what I am talking about! :)
I tried this from the terminal with no effect:
`set env QT_SCALE_FACTOR=1
`
Not sure if that's what I am supposed to do or not?
Cheers,
Jon

fieldOfView, 2018-04-19:
@thopiekar, any ideas? I don't know the first thing about fedora.

thopiekar, 2018-04-19:
Hey @fieldOfView :)

What do you mean by "first thing"?

Afaik, scailing should work. I did that already in the past and only a hand full of widgets were not scaled. But at least Cura ist then usable again.
To be honest I never use "set env". Don't know what this specifically does, but try to use "export" instead.

Cheers! :wink:

fieldOfView, 2018-04-19:
(@thopiekar: "I don't know the first thing about X" means "I don't know anything about X")

@JonHeron: how do you normally start Cura? From a commandline or with an icon? If you just set an environment variable from a terminal, it could be that that environment variable is only set in that terminal session, so if you then use some icon in your desktop manager to launch Cura that environment variable will not be set for that session.

thopiekar, 2018-04-19:
@fieldOfView : Haha, alright. I'm a Ubuntu user and sometimes using Debian. Personally I would say there are many differences between the distros. The only thing I see is the choice of how software is maintained.

Some things are still unclear to me:
* Which desktop environment is being used?
* Are you using Xorg or Wayland ( some settings are still not synched between both systems)
* Are you using the official aka supported aka holly Ultimaker AppImage? Or the software packages, which come with your distro?

Thanks :grinning:

JonHeron, 2018-04-19:
Sorry for leaving out the important details:
Using Wayland
Gnome 3.26.2
Using Cura from the official fedora software repo
I tried running it from the command line after setting the env too, with the same results.
There is some debug info I see in the log from terminal:
`[jonnyslaptop@jonnyslaptop-localdomain ~] $ cura
[(140257942676800)-MainThread] UM.Resources.__initializeStoragePaths [351]: Initializing storage paths
[(140257942676800)-MainThread] UM.Resources.__initializeStoragePaths [362]: Config storage path is /home/jonnyslaptop/.config/cura/3.2
[(140257942676800)-MainThread] UM.Resources.__initializeStoragePaths [370]: Data storage path is /home/jonnyslaptop/.local/share/cura/3.2
[(140257942676800)-MainThread] UM.Resources.__initializeStoragePaths [382]: Cache storage path is /home/jonnyslaptop/.cache/cura/3.2
[(140257942676800)-MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
[(140257942676800)-MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile
[(140257942676800)-MainThread] UM.Qt.QtApplication.__init__ [88]: Detected most suitable OpenGL context version: 4.1 Core profile
[(140257942676800)-MainThread] UM.Qt.QtApplication.__init__ [104]: Command line arguments: {'debug': False, 'trigger_early_crash': False, 'external-backend': False, 'headless': False, 'disable-textures': False, 'qmljsdebugger': None, 'file': [], 'single_instance': False}
2018-04-19 07:44:53,429 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin ConsoleLogger
2018-04-19 07:44:53,478 - INFO - [(140257942676800)-MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [65]: Found CuraEngine at: /usr/bin/CuraEngine
2018-04-19 07:44:53,479 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraEngineBackend
2018-04-19 07:44:53,482 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2018-04-19 07:44:53,486 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2018-04-19 07:44:53,486 - WARNING - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [200]: Plugin ConsoleLogger was already loaded
2018-04-19 07:44:53,494 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLReader
2018-04-19 07:44:53,496 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJReader
2018-04-19 07:44:53,498 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLWriter
2018-04-19 07:44:53,500 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJWriter
2018-04-19 07:44:53,549 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin UpdateChecker
2018-04-19 07:44:53,555 - INFO - [(140256764745472)-Thread-1] UpdateChecker.UpdateCheckerJob.run [34]: Checking for new version of cura
2018-04-19 07:44:53,558 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2018-04-19 07:44:53,562 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2018-04-19 07:44:53,567 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2018-04-19 07:44:53,570 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2018-04-19 07:44:53,574 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2018-04-19 07:44:53,577 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2018-04-19 07:44:53,579 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalContainerProvider
2018-04-19 07:44:53,581 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2018-04-19 07:44:53,584 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeProfileReader
2018-04-19 07:44:53,678 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2018-04-19 07:44:53,682 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin PostProcessingPlugin
2018-04-19 07:44:53,685 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
2018-04-19 07:44:53,700 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin 3MFWriter
2018-04-19 07:44:53,704 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeReader
2018-04-19 07:44:53,713 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimulationView
2018-04-19 07:44:53,716 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
2018-04-19 07:44:53,719 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin UserAgreementPlugin
2018-04-19 07:44:53,721 - INFO - [(140257942676800)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeWriter
[... cut ...]
2018-04-19 07:45:02,332 - DEBUG - [(140257942676800)-MainThread] CuraEngineBackend.CuraEngineBackend._terminate [262]: Attempting to kill the engine process
2018-04-19 07:45:02,333 - DEBUG - [(140257942676800)-MainThread] CuraEngineBackend.CuraEngineBackend._terminate [268]: Killing engine process
2018-04-19 07:45:02,334 - DEBUG - [(140257942676800)-MainThread] CuraEngineBackend.CuraEngineBackend._terminate [271]: Engine process is killed. Received return code -15
`
`[jonnyslaptop@jonnyslaptop-localdomain ~] $ cura --version
cura 3.2.1
`
Cheers,
Jon

fieldOfView, 2018-04-19:
Try starting Cura with ```QT_SCALE_FACTOR=1 cura```, but also try other values (eg ```QT_SCALE_FACTOR=0.5 cura```)

thopiekar, 2018-04-19:
And please try to reproduce while using Xorg. 🙂

JonHeron, 2018-04-19:
@fieldOfView Using a scale factor less then 1 breaks the UI. Using 1 or above does not fix the issue.
@thopiekar  same issue confirmed with wayland and xorg.
Cheers,
Jon

fieldOfView, 2018-04-19:
What do you mean by "breaks the UI"? Please be more specific (another screenshot perhaps?). Also when using a value > 1, does that change anything (ie: do the fonts get even bigger)?

JonHeron, 2018-04-19:
Here it is @ .5
http://theherons.ca/album/var/albums/3D_Printing/Screenshot%20from%202018-04-19%2008-59-34.png?m=1524142862
http://theherons.ca/album/var/albums/3D_Printing/Screenshot%20from%202018-04-19%2008-59-46.png?m=1524142863
This is the same for anything <1.
Jon

ianpaschal, 2018-04-26:
I'm not a linux user so I can't do much testing on my own, that's why I'm slightly deferring to @thopiekar's comments.

Tagging our Cura devs who use linux: @jackha @Ghostkeeper, although I'm sure this doesn't recreate for them or they wouldn't get any dev work done! @nallath works on CuraConnect but he uses Fedora so maybe he can help.

I'm going to label this as a bug but since it's highly unlikely any of us will be able to recreate the bug, I would not hold your breath. Sorry. ☹️

JonHeron, 2018-04-26:
Thanks for your efforts!
Cheers,
Jon

jackha, 2018-04-26:
Can't check or reproduce it on my Ubuntu with Unity...

ianpaschal, 2018-04-26:
Bad luck Jon, we use PyQt so the only C++ code is the CuraEngine (slicing backend).

JonHeron, 2018-05-23:
Using the latest appImage 3.3.1 and the latest F28..... the bug is dead!
Looking forward to trying it out (when I get home from this business trip :) )
   
Cheers!
Jon

diegopradogesto, 2018-05-23:
Hey, glad to hear! Thanks for the info.

closed 2018-05-23 by diegopradogesto