Ultimaker/Cura #3680: Cura Plugin

tmirante, 2018-04-17:
According to what signaled
" I think the plugin that will be most helpful to get you started if you want to contribute such a plugin is the BedLevelMachineAction which can be found here:
https://github.com/Ultimaker/Cura/tree/master/plugins/UltimakerMachineActions"

and i found that UltimakerMachineActions is installed in Cura 3.2.1 . but when i go looking for BedLevelMachineAction in the SCRIPT to activate i not find it. What i must do to find ?

[image]

fieldOfView, 2018-04-17:
There is a big difference between a "plugin" and a "script".

Cura consists of "plugins" that provide its functionality. Everything from loading STL files to talking with the CuraEngine backend, from USB Printing to the tools in the toolbar are all plugins. Some plugins are MachineActions, such as the UltimakerMachineActions. To see these in action, please add a new Ultimaker Original to your Cura setup. They also manifest on the Printers pane of the preferences.

PostProcessing, by itself also a "plugin", has the ability to load "scripts". These scripts are what you see in the dropdown in the screenshot.

fieldOfView, 2018-04-17:
Note that because you opened a new issue, nobody knows what you are talking about. I think you quote me, but all the others here have no idea who pointed you to UltimakerMachineActions and why.

tmirante, 2018-04-17:
excuse me , i am new here . your answer means that if i add ultimaker original printer i will see addon  for Bedlevel i suppose so i can use it on my prusa printer ?

tmirante, 2018-04-17:
i will test it , but let me understand how i can have the same function available without installing ultimaker printer ?  can i use it and adding in actual installation ?

tmirante, 2018-04-18:
my effort to begin development is to get BedLevelMachineAction and convert it to be acceped as a plugin in the script directory. after i will modify it to adapt to requirements for all printers asking for parameters. so first step is to let it accept as a plugin. i changed files and charge in script drectory but i not see it listed on cura. May help please there will be a step i lost 

[TMBedLevelMachineAction.zip](https://github.com/Ultimaker/Cura/files/1923677/TMBedLevelMachineAction.zip)

tmirante, 2018-04-19:
no answers about this request, nobody may help to check why my plugin is not accepted from cura ?

fieldOfView, 2018-04-19:
I started my first answer above with the following statement:

> There is a big difference between a "plugin" and a "script".

I continued to explain the difference, yet you seem to want to make a script out of a plugin, which is conceptually impossible. I don't know how I can help you.

tmirante, 2018-04-19:
I don't want to make a script , i want to do a PLUGIN ( my post was corrected) . 
to do it i got BedLevelMachineAction plugin and i transformed internal _init_ file and others to create a NEW PLUGIN called TMBedLevelMachineAction. After i loaded it in PLUGIN directory and i was thinking to be able to install it with the option to install a new plugin. but i find that cura is looking for *.curaplugin files or *.umplugin files and i don't know how to create them as my files are simple .py  json and qml.  What i must do ?  i have the last version py compiler with uranium installed if required   .

ianpaschal, 2018-04-19:
You can find the answers to that [here](https://github.com/Ultimaker/Uranium/wiki/Creating-plugins).