Ultimaker/Cura #3681: Submitting new slicing parameters for a printer - is it too late for Cura 3.3?

filipgoc, 2018-04-17:
**Printer**

IMADE3D JellyBOX

**Question**

Is it too late to submit new slicing presets? This does in no way affect the rest of Cura; it's only affecting the users of one particular printer. 

If it's not too late, could I get a rough cut-off? Which week or so i'll be frozen?

THANKS.

fieldOfView, 2018-04-17:
Generally, when a Cura version goes in beta it is “feature frozen”; only important bugfixes are accepted into a beta.

filipgoc, 2018-04-17:
Yeah. I was afraid of that @fieldOfView . I hoped that preset could be exempt since they are not technically a feature.

ChrisTerBeke, 2018-04-17:
Unfortunately we usually don't update settings this close to a release. As you might have seen in some of the latest commits, we're actually working on decoupling the resources from the main codebase so that we can update these without having to release Cura. So in the (near) future this will improve!

closed 2018-04-17 by ChrisTerBeke