Ultimaker/Cura #3682: [3.2.1] Make Overhang Printable custom angle selection

superjamie, 2018-04-18:
The **Make Overhang Printable** option in Experimental is great!

For those not familiar, it takes an unsupported overhang (eg: on a head model, the underside of the chin is an unsupported overhang) and extends the model downwards and inwards to create a printable support which becomes part of the model.

From casual observation this appears to extend downwards at an angle of 45 degrees or 60 degrees.

Different printers/settings are capable of different overhangs, so it would be good to have the added overhang angle configurable.

For example, given the following overhang:

    -----,
         |
         |
         |
         |

An overhang angle of 45 degrees would add material like so:

    -----,
    \    |
     \   |
      \  |
       \ |

But an overhang angle of say 75 degrees would add material like so:

    -----,
     `.  |
       `.|
         |
         |

It would be up to the user to explore the limits of the most extreme angle their printer/settings are capable of.

Please add tag **Type:Improvement**

BagelOrb, 2018-04-19:
I think this is already implemented. It's just the Overhang Angle (if I remember correctly).

Enable support, edit overhang angle and disable support again.

superjamie, 2018-04-21:
I tried this with a few overhang models but it didn't seem to make any difference. Do you have a suggested model where the change is easy to see?

BagelOrb, 2018-04-21:
There already is a seting for it and it is called "Maximum Model Angle".

closed 2018-04-21 by BagelOrb

marcottt, 2019-02-19:
i'm on cura 4.0 but unable to use this option, i enable voice and put overhang angle to 45 but support are still the same.  I need some trick ?

Ghostkeeper, 2019-02-22:
You need to enable the setting "Make Overhang Printable" and then set the Maximum Overhang Angle setting underneath it to your desired angle.