Ultimaker/Cura #3683: [3.3.0 beta] Pause/Resume doesn't work.

kirilledelman, 2018-04-18:
Paused print around 70% to fix stuck filament. Clicked Resume, nothing happens.

Additionally, move axis buttons don't work in paused state. Ideally, I'd like to pause, move Z up, do stuff to extruder, move Z down, resume.

Appesteijn, 2018-04-20:
What printer do you have? We could use some more info.

ianpaschal, 2018-04-20:
Yes, @Appesteijn is correct. Please use the issue template provided when opening an issue. It takes a minute tops to explain what you were doing and what results you got, and without it the possibility that we can diagnose the problem and fix it is basically zero. Please fill it out or we'll have to close the issue.

kirilledelman, 2018-04-21:
Fair enough:
3.3.0 Cura beta 
Mac OS - High SIerra
Custom FDM printer - Marlin 1.8, dual extruder
[Left_box.3mf.zip](https://github.com/Ultimaker/Cura/files/1934458/Left_box.3mf.zip)

adrianmihalko, 2018-04-27:
Same problem with latest cura and Creality Ender 3.

Ghostkeeper, 2018-05-25:
I can't reproduce the bug. It properly resumes when I click Resume. I tested this using an Ultimaker 2 as I don't have the Creality Ender 3 available here.

The way that resuming works is a bit weird. While printing, it normally sends the next command after the printer responds to the previous command with "ok". When you pause, it sees the "ok" but then sends nothing any more. When you resume, it will no longer ignore the next "ok" message so then it continues with sending the next g-code command. But the printer only responds to "ok" after getting g-code. The reason that this works is because every 2 seconds we poll the temperature, which should also give a "ok" message.

So my question is: Does the nozzle temperature indicate correctly in the Cura monitor page? Because if not, then the problem is really that your printer doesn't respond with "ok" to our regular temperature polling.

In the meanwhile I'll see what I can do to make the code a bit more logical and stable there.

closed 2019-01-07 by nallath

krikru, 2021-06-28:
I had the same issue with the dog test model gcode that came on the microSD card and an Ender-3 I bought new a few weeks ago, so this seems to be purely a Creality issue. When I paused the print, the printer head wouldn't return to home, and when I pressed resume, the nozzle and bed wouldn't reheat again. As a workaround, if I turn off and turn on the machine and then choose resume, the bed and nozzle reheats, the printer head returns to home, and then printing is resumes.