Ultimaker/Cura #3686: ModuleNotFound UM.ConfigurationErrorMessage

g3n35i5, 2018-04-19:
Cura 3.1.0 ArchLinux 64 Bit
Cura can not load the module ConfigurationErrorMessage from Ultimaker (UM).
Trace:

```
Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3.6/site-packages/cura/CuraApplication.py", line 56, in <module>
    from cura.Machines.Models.BuildPlateModel import BuildPlateModel
  File "/usr/lib/python3.6/site-packages/cura/Machines/Models/BuildPlateModel.py", line 11, in <module>
    from cura.Machines.VariantManager import VariantType
  File "/usr/lib/python3.6/site-packages/cura/Machines/VariantManager.py", line 8, in <module>
    from UM.ConfigurationErrorMessage import ConfigurationErrorMessage
ModuleNotFoundError: No module named 'UM.ConfigurationErrorMessage'

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/cura", line 128, in <module>
    import cura.CuraApplication
  File "/usr/lib/python3.6/site-packages/cura/CuraApplication.py", line 56, in <module>
    from cura.Machines.Models.BuildPlateModel import BuildPlateModel
  File "/usr/lib/python3.6/site-packages/cura/Machines/Models/BuildPlateModel.py", line 11, in <module>
    from cura.Machines.VariantManager import VariantType
  File "/usr/lib/python3.6/site-packages/cura/Machines/VariantManager.py", line 8, in <module>
    from UM.ConfigurationErrorMessage import ConfigurationErrorMessage
ModuleNotFoundError: No module named 'UM.ConfigurationErrorMessage'
```

fieldOfView, 2018-04-19:
UM.ConfigurationErrorMessage was recently added to master. It looks like you are somehow running the master branch of Cura with an older branch of Uranium. Both branches should match.

diegopradogesto, 2018-04-20:
Yes, that's probably the reason. Thanks @fieldOfView.
I will close this issue but be free to open it again if it doesn't work.

closed 2018-04-20 by diegopradogesto

g3n35i5, 2018-04-21:
Thank you @fieldOfView for your help, it worked!

Best regards
g3n35i5