Ultimaker/Cura #3692: Print time won't display correctly in German

10MINT, 2018-04-21:
**Application Version**
3.2.1

**Platform**
Windows 10

**Display Driver**
NVIDIA GeForce GTX 1050 Ti @ 1920 x 1080 pixels
driver nvldumdx.dll in version 23.21.13.8871

**Printer**
Tevo Tarantula

**Steps to Reproduce**
1. Set lanuage to German
2. Slice any file

**Actual Results**
The print time won't display correctly (You can't read the time because the minutes are cut

**Expected results**
The print time should be readable

**Additional Information**
[image]

Appesteijn, 2018-04-21:
Do you have a suggestion on how we could shorten the text on the button?

fieldOfView, 2018-04-22:
I think the “wechseldatentrager” was changed to “datentrager” recently for this reason. “Stunden” could als be shortened to St.

10MINT, 2018-04-22:
What @fieldOfView suggested could solve the issue

Ghostkeeper, 2018-05-11:
The "datentrager" translation was already applied. I've now also applied the other suggestion with these two commits:

* https://github.com/Ultimaker/Cura/commit/8e3e5f50b4d3d7fb5263fe1a90ca66f3ae7e8943
* https://github.com/Ultimaker/Uranium/commit/63b6da99c3447509810b6a862a097f335b5d37a1

It should be easier to read next release.

commit 8e3e5f50 (2018-05-11): Improve translation of short time format

It wasn't short enough.

Fixes #3692.
closed 2018-05-11 by Ghostkeeper