Ultimaker/Cura #3696: [REQUEST] Support Initial Line Distance

Dabihu, 2018-04-22:
Support adhesion could improve easily and efficiently, when different "Support Line Distance" could set for the initial layer.

It could be very useful, when screw-nut or small supports needed, but during printing the head easily knock out the support. Or like this closed issue: #679

ianpaschal, 2018-04-25:
Could you explain how it why it would be different for higher layers? I agree it's possible for the print head to knock out the support but don't understand why you'd only want to make it dense for the first layer and not layer 2 and up.

Dabihu, 2018-04-27:
With default setting the support first layer is only some lines on the bed:
[image]
Higher these lines are enough, but sometimes the head grab the support from bed at this point:
[image]
The full height shown like this:
[image]
If the "Support Line Distance" is 0.4 mm:
[image]
Heigher:
[image]
Another example, what I printed 3 times, because the model is this:
[image]
First layer:
[image]
This is layer 78, around this height when the head is 15 mm above the full support knocked of from the bed, because head is traveling on every layer diagonal on the top of the support after printed it:
[image]
If I used brim the head could not knock the support.
Another solution could be retract and z-hop, but this had some time penalty and traces on the printed surface.

AbeFM, 2018-05-13:
I guess you could make gradual supports both directions. 

This seems similar to topic on support build plate adhesion.

Dabihu, 2018-05-13:
Similar request: #3759

AbeFM: gradual support doesn't work on build plate or I could not found what you were thinking.

smartavionics, 2018-05-16:
Hello @Dabihu , this isn't a comment regarding your request but simply pointing out that supports can often be avoided completely for features like nut traps, etc. Just put a single layer over the hole and drill it out later. Not only is it much quicker to print but the resulting quality is better.

[image]

[image]

[image]

I also use the same technique for horizontal holes and that again leads to improved print quality. In this example, I am showing how the 4 horizontal holes in the part have a thin web through the middle which gets drilled out later. By doing this, it means that the outer wall of the whole part can be printed as one which leads to a better quality.

[image]

Dabihu, 2018-05-20:
Hello @smartavionics, your ideas are good and sometimes very useful. Thank you!
But the main point of 3D printing that you could print thinks that don't need post machining in most cases.
The main problem, that the support stick only with some lines to the bed and not always possible eliminate the support.

smartavionics, 2018-05-20:
>Hello @smartavionics, your ideas are good and sometimes very useful. Thank you!
But the main point of 3D printing that you could print thinks that don't need post machining in most cases.

Yes (and no!) it depends on what you are wanting to achieve. Fastest results or best finish. You can't always get both.

> The main problem, that the support stick only with some lines to the bed and not always possible eliminate the support.

Yes, I can see that could be a problem. I think several people have been asking about this feature recently.

jackha, 2018-05-31:
I discussed this in the team and we're not going to introduce new settings. Current there are plenty of ways to fix this with the current options available:
- The options mentioned above
- Avoid Supports When Traveling  <-- introduced 2 months ago ^^
- Sounds dumb, but higher support density?

closed 2018-05-31 by jackha

AbeFM, 2018-05-31:
I don't want to beat it to death, but a soluble supports piece is often to
achieve here surface finish - a brim can leave an ugly edge.

Just leaving on floor settings seems like very little extra work?

Now I just lift the part of the bed entirely. Worse adhesion, but then I
can be aggressive on supports.

Thanks for the "avoid supports" info, I'll be trying that!! Still, you
can't avoid them when printing on top/making them taller.

And thanks for the consideration.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Dabihu, 2018-05-31:
@jackha understood your point, but I didn't get a good solution for the problem:
- Options above doesn't solve the weak support adhesion on surface
- "Avoid Supports When Traveling" doesn't found in Cura 3.3.1. Just avoid printed parts and this doesn't care for support structures If I remember correctly. (Travels on the top of them.) I'll look into it in a couple of days.
- Higher support density is good, but no point to do a high density support only for a good adhesion. Only the first layer must have a much higher density.

Dabihu, 2018-06-02:
@jackha I tested my memories. As I thought. "Avoid Printed Parts When Traveling"  doesn't avoid printed supports. It's print the support, and after that it goes across the top of them in high speed. Blue paralel lines in the X directions on every layer.
[image]
Attached a simple STL file and the Cura 3.3.1 generated gcode in a zip file.
[SupportTest.zip](https://github.com/Ultimaker/Cura/files/2064810/SupportTest.zip)

jackha, 2018-06-04:
Ahh!! Sorry, "Avoid Supports When Traveling" is introduced in the upcoming beta :-)
I didn't test it, but it should help not bumping supports that have a weak adhesion.
Indeed a higher support density only for better adhesion is not ideal, but since there are quite a lot solutions already (although not all optimal), we've decided to focus on other features and bugs. I'm putting my hopes on Avoid Supports When Traveling :-)

Dabihu, 2018-06-04:
Thank you! :)
I hope too that "Avoid Supports When Traveling" will be a solution to this problem.

AbeFM, 2018-06-04:
Poorly adhered supports effect more than supports falling off. Printing in an unstable support leads to bad surfaces in the area printer above it.

Minimum contact patch might make sense. Still a floor or raft at JUST the base of the supports would help.

Looking forward to trying avoid supports