Ultimaker/Cura #3698: Zig-zag skin pattern sometimes has spurious lines at ends

smartavionics, 2018-04-22:
**Application Version**

Master branch
**Platform**

Linux
**Qt**

5.8.0
**PyQt**

5.8.2
**Display Driver**

OpenGL Version:  4.5 (Core Profile) Mesa 17.2.8
**Printer**

Kossel XL
**Steps to Reproduce**

Load model, slice using zigzag skin pattern, view layers.

**Actual Results**

Zig-zag skin sometimes has extra line segments.

**Expected results**

Zig-zag skin should not have extra line segments.

**Additional Information**

This image shows an example. In the top left corner is an extra line.

[image]

Before that line is output, it looks correct:

[image]

smartavionics, 2018-04-23:
The extra lines are fairly harmless except for the fact that they require an extra travel move. I don't think this used to happen so something has changed in the zig-zag department.

Ghostkeeper, 2018-05-11:
It seems to have something to do with skins, since I can't reproduce this with infill but I can with skin.

jackha, 2018-07-03:
We discussed it and we're not going to fix this.

smartavionics, 2018-07-03:
That's OK because I have a PR (https://github.com/Ultimaker/CuraEngine/pull/792) that can help reduce the effect of these spurious lines.

closed 2018-07-03 by smartavionics