Ultimaker/Cura #3700: Spiralize Outer Contour Mode and Lift Head (Cooling) should be mutually exclusive

akf-was-here, 2018-04-24:
**Application Version**
 3.2.0 / ALL

**Platform**
 Windows 10

**Qt**
 Your version

**PyQt**
 Your version

**Display Driver**
 Nvidia 391.45

**Printer**
 Any

**Steps to Reproduce**
 Spiralize Outer Contour Mode and Lift Head (Cooling) should be mutually exclusive.

**Actual Results**
 Lifting head during spiral mode breaks spiral mode

**Expected results**
 Enabling spiral mode should disable Lift Head.

**Additional Information**

smartavionics, 2018-04-24:
I agree, I've coded a fix in https://github.com/Ultimaker/CuraEngine/pull/751, it's up to the Cura devs now to decide whether they like that or not. Cheers!

Ghostkeeper, 2018-05-11:
I'd rather not prevent this entirely since there is no technical limitation to it. According to Cura's conventions, if we think that some setting would give bad print results, we should give a warning colour. It should still be possible to lift the head if the user does think it's better for his print. I guess it could still be viable if you're printing with a very large nozzle size and printing a thin pillar, since it will sag from all the heat.

The reason that this warning is not given yet is that this requires an implementation for warning values with boolean settings. We don't have that yet.

akf-was-here, 2018-05-11:
How about linking the two so that the "Lift Head" feature is switched off
with enabling "Spiralize Outer Contour Mode" but that nothing prevents
someone from switching "Lift Head" back on?

I think it's a fairly aberrant case where you'd want both on at the same
time as lifting the head is going to kill almost all spiral prints.  I know
it has killed several of mine since Spiral Mode prints are not my normal
mode of printing and lift head is an awesome feature to have on normally.

jackha, 2018-07-02:
Devs: CURA-5467

Ghostkeeper, 2018-08-21:
Our project manager removed this from our planning. It was not deemed important enough to spend time on it now, so it must be deferred.