Ultimaker/Cura #3701: Different Layer Heights (normal/initial) cause wrong support

pappy15, 2018-04-24:
**Application Version**
3.2.1

**Platform**
Windows 10

**Steps to Reproduce**
Layer Height 0.2
Initial Layer Height 0.1
Generate Support

**Actual Results**
The model moves one layer up, so the initial layer only consists of support and brim. If the initial layer height is set to 0.2 like normal layer height, model and support start together at layer one.

**Expected results**
Model and support should start together at layer one, like it is with same layer height and initial layer height.

layer height 0.2 | initial layer height 0.1
[image]

layer height 0.2 | initial layer height 0.2
[image]

pappy15, 2018-04-25:
I updated to cura 3.3 and the problem seems to be fixed now. Thank you.

smartavionics, 2018-04-25:
OK, great, please close the issue if you are happy.

closed 2018-04-25 by pappy15