Ultimaker/Cura #3702: [3.3.0] Crash on startup if installed on D:\ rather than C:\

jackjt8, 2018-04-24:
**Application Version**
Cura 3.3.0

**Platform**
Windows-10-10.0.16299
(64 bit)

**Qt**
default 5.8.0

**PyQt**
default 5.8.2

**Display Driver**
Intel HD Graphics 630 - 23.20.16.4877
(Not yet initialized)

**Printer**
N/A

**Steps to Reproduce**
Switch install directory from C:\ to D:\. Run Cura.

**Actual Results**
Cura experiences a fatal error in PluginsModel.py

**Expected results**
Like past Cura versions the install drive should not matter.

**Additional Information**
[Logs.txt](https://github.com/Ultimaker/Cura/files/1943177/Logs.txt)
[traceback.txt](https://github.com/Ultimaker/Cura/files/1943178/traceback.txt)
[image]
[image]

I have already deleted Cura data in both AppData\Local\ and AppData\Roaming\. I have tried installing to both C:\ and D:\ multiple times and only installing to C:\ works. Cura 3.2.1 does not care which drive it is installed on.

TheBennyJ, 2018-04-24:
I am running into the same issue, installing on the C:\ also resolved the issue for me on Windows 10. My previous install of 3.2.1 was on the E:\ drive.

MB3Y, 2018-04-24:
Another confirm of the above.

nzidol, 2018-04-25:
No can't install to c-drive either, software was original installed on d-drive under Windows 10....
```
Traceback (most recent call last):
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\PluginsModel.py", line 44, in __init__
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\PluginsModel.py", line 83, in _update
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\PluginRegistry.py", line 322, in isBundledPlugin
  File "ntpath.py", line 632, in commonpath
ValueError: Paths don't have the same drive
```

Appesteijn, 2018-04-25:
Could you try this build: https://stack.mybacklog.info/s/OdSi5xGBeTjwNAf ?

LipuFei, 2018-04-25:
Hi all, FYI, we have made a fix for this issue: https://github.com/Ultimaker/Uranium/commit/5c4ddb0dacabb7b56194d62ea62fb54dea1d12cf

closed 2018-04-25 by LipuFei