Ultimaker/Cura #3704: Cura 3.3.0  crash  on OS-X

bobcamp, 2018-04-24:
Hi

Running Mac OS-X, latest version with all patches up to date.  Version Cura 3.2.1 running fine into UM 3 ext and UM3. 

Download version 3.3.0 from UM web site. Extract to disk and launch. 

Programe makes it to some sort of dialog that then crashes before it can be read. Uninstall / reboot / reinstall does no good. Roll back to 3.2.1 and it works fine. Log as follows per request from UM.

[cura.log](https://github.com/Ultimaker/Cura/files/1943833/cura.log)

bobcamp, 2018-04-24:
Digging through the log:

2018-04-24 09:37:07,917 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2018-04-24 09:37:07,919 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2018-04-24 09:37:08,079 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [133]: Need to check for errors again. Discard the current progress and reschedule a check.

Seems to be the point that it ends ..... good old Python ...

Bob

MrWetzelEdTech, 2018-04-24:
Here is my log....
[cura.log](https://github.com/Ultimaker/Cura/files/1944024/cura.log)

bobcamp, 2018-04-24:
This could easily be the same thing as happens on Windows. If so feel free to delete / merge this whole issue

adrianmihalko, 2018-04-24:
I have same problem. I downloaded cura 3.3.0/Mac and when I try to launch it I see for a 1-2 sec the license agreement popup and it's crashing.

adrianmihalko, 2018-04-24:
3.2.1 confirmed to works fine.

jazzychad, 2018-04-25:
Same issue here. Installed 3.3.0 on Mac 10.13.3. I see license agreement for a second then crash.

```
2018-04-24 21:21:08,884 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2018-04-24 21:21:08,885 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2018-04-24 21:21:08,888 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2018-04-24 21:21:08,890 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2018-04-24 21:21:09,187 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [133]: Need to check for errors again. Discard the current progress and reschedule a check.
```

Ellecross, 2018-04-25:
Hey @bobcamp @adrianmihalko  and @jazzychad, thanks for reporting this. 

Did you guys happen to backup your configuration folders from 3.2.1 in the /application support/cura directory? If so can you compress and attach it here? This will help us identify what caused the crash for you in Mac.

adrianmihalko, 2018-04-25:
First I had installed 2.6, then 3.3, but because it crashed I installed 3.2.1.

My cura dir zipped:

https://hopp.mihalko.eu/2HNv7Ni

Ellecross, 2018-04-25:
@adrianmihalko I found out what was causing yours to crash and way to workaround it without losing your stuff. 

Please delete the folder highlighted in this image: 
[image]

If you remove those and start 3.3 again with only 2.6 and 3.2 in the Cura folder, then you should be good. I see your printers and the custom profiles as well from 3.2. 

What happened to you was quite odd. The folders highlighted, except the 3.3, should all go under the cura version folder. They should not be in the root Cura folder. I don't know how that happened. :/ 

Normally, your roaming/Cura folder should just look like this: 
[image]

and then within each version folder, it should contain the rest: 
[image]

Let me know if this helps you!

Ellecross, 2018-04-25:
@bobcamp @MrWetzelEdTech and @jazzychad let me know if the same workaround (my comment above) will work for you. Otherwise, send me the configuration folder then I'll check. Thanks

LipuFei, 2018-04-25:
For Dev, JIRA ticket: CURA-5281

LipuFei, 2018-04-25:
Hi adrianmihalko, I have checked your cura.zip and it looks like a problem in version upgrade. I'm working on that now. And @Ellecross @bobcamp @MrWetzelEdTech , could you also upload your cura directory so we can check what went wrong? Thank you!

MrWetzelEdTech, 2018-04-25:
@LipuFei I am away from my computer this morning. I will check it later today and confirm accordingly. I am on a Mac; will the process be any different other than the slight change in the directory path?

jackha, 2018-04-25:
The problem with the settings dir from @adrianmihalko is fixed in the first upcoming version.

Fix: https://github.com/Ultimaker/Cura/commit/c4994918c8e0750c5dd010a8617a767e32187189

Ellecross, 2018-04-25:
@MrWetzelEdTech For mac, it is \Library\Application Support\cura\. Please get the 3.2 folder so we can see what happened. Thanks!

bobcamp, 2018-04-25:
Hi 

No luck with the delete of the "extra" directories. Also no luck uploading the zip file. It's over 10 Meg and I get a "Yowza" message. Even the directory structure without the spurious files is 32 meg after compression. 

Bob

bobcamp, 2018-04-25:
Hi 

Pardon the chop job .... this is the 3.2 directory that is under the 3.3 directory

[3.2.zip](https://github.com/Ultimaker/Cura/files/1947014/3.2.zip)

bobcamp, 2018-04-25:
This is the 3.1 directory

[3.1.zip](https://github.com/Ultimaker/Cura/files/1947029/3.1.zip)

bobcamp, 2018-04-25:
This is the 3.3 directory, the log file from 3.2 was deleted to get things to fit. I'll  go dig that up next

[3.3.zip](https://github.com/Ultimaker/Cura/files/1947038/3.3.zip)

bobcamp, 2018-04-25:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/1947054/cura.log.zip)

That should be it. To re-assemble: 

All the files except the last one go into.a 3.3 folder. The log file goes into 3.2 under the 3.3. I hope that makes sense.

Bob

bobcamp, 2018-04-25:
If there is an alternate upload process, I can feed you all of the zips of all of this. It's all saved.

Bob

bobcamp, 2018-04-25:
Did the spring that popped out of the extruder wind up in any of those zip files? I don't see it on the floor anywhere around here :) 

Bob

LipuFei, 2018-04-25:
Should be fixed in df1e15b1e51fd7b4af4da84b52b7b35542175ea4

closed 2018-04-25 by LipuFei

Ellecross, 2018-04-25:
Thanks for providing the files @bobcamp, we have a build coming soon that will have this fixed.

adrianmihalko, 2018-04-25:
Thank you guys, I just received my printer so I don't have any important settings.

rm -rf ~/Library/Application Support/cura/ did the job.

LipuFei, 2018-04-25:
If you have something to keep, remove the `3.3` folder in `~/Library/Application Support/cura`. Otherwise, you can always remove the whole `~/Library/Application Support/cura` to have a clean start.

bobcamp, 2018-04-25:
Ok 3.3.1 is out. It *does* load properly. After it loads, it's convinced that the only printer is my Prusa I3. All of the settings for the UM3 are gone. Not the same bug so indeed this one is fixes. We just have another bug to track down. I note that there is a ticket out already on "loose settings". I *assume* the rest of my problems should move over there?

bobcamp, 2018-04-25:
If I nuke all of the 3.3 stuff ( the 3.3 folder and all the "extras") and then load 3.3.1, all is well. It's a bug in a bug in a bug. Probably not worth digging very far into.