Ultimaker/Cura #3705: [3.3.0] Crashes on startup after upgrade from 3.2.1

Snowflake6, 2018-04-24:
**Application Version**
3.3.0

**Platform**
Intel (R) Core(TM) i7-6700 CPU @ 3.40 GHz
16.0 GB Installed RAM
Windows 10 Pro V 1709 Build 16299.192

**Qt**
From Ultimaker Website

**PyQt**
From Ultimaker Website

**Display Driver**
NVIDIA Quattro K2200
2016-10-21 V21.21.13.7563

**Printer**
Ultimaker 2+ selected during install.

**Steps to Reproduce**
Install Cura 3.3.0 over top of 3.2.1.
When asked, remove 3.2.1 during install.

**Actual Results**
When Cura starts, release notes box comes up, then disappears before you can close it.  Cura executable dies and nothing further happens.  Nothing visible in Task Manager.

**Expected results**
Should work like previous versions.

**Additional Information**
Installed in C:\Program Files\Ultimaker Cura 3.3, previous versions were installed on C drive as well.

Ellecross, 2018-04-25:
Hey @Snowflake6 sorry you're getting this crash. But thanks for reporting this issue. 
Will you be able to compress and attach your configuration folder from 3.2.1 here? This will help me identify what caused your crash. It should be in /Appdata/Roaming/Cura/3.2 .. 

What @matmat07 is saying is also true, should you have an octoprint plugin. It's a known issue now and we're working to have it fixed.

alwashe, 2018-04-25:
i could not select anything after installation, crashes right after start. same configuration as above.

Install Cura 3.3.0 over top of 3.2.1

Snowflake6, 2018-04-25:
@Ellecross, I can look for the config when I get in to the office.  I have removed 3.3.0 and re-installed 3.2.1 to get myself working for now.  I don't use Octoprint as far as I know, at least I didn't install it separately.  If it came with the default Cura package that's the only way it could get there.

Appesteijn, 2018-04-25:
Please try 3.3.1: https://ultimaker.com/en/products/ultimaker-cura-software

Snowflake6, 2018-04-25:
3.3.1 confirmed to work properly.  I will assume there's no point in posting config files from 3.2.1 now.

closed 2018-04-25 by Snowflake6