Ultimaker/Cura #3706: Cura3.3 stable losses all previous user data on instal and launch

drdavelight, 2018-04-24:
**Application Version**

3.3 Stable
**Platform**

Upon install and launch, ALL previous user data is lost all printers, materials, settings, plugins, viewer settings and even quality constraints
**Qt**

**PyQt**

**Display Driver**

Intel Core i7
**Printer**

ALL were GONE we HAD 4 but all missing except still in 3.2.1 when I open that version
**Steps to Reproduce**

Removed and reinstalled happened again IDENTICALLY seems to not know where to find the previous files to install them upon loading
**Actual Results**

**Expected results**

ALl prev oous setting options and printers should have transferred over to the new version.  This was common in version prior to 2.x and was supposedly fixed in 2.x and has been working fine as I have routinely upgraded with no difficulties
**Additional Information**

Appesteijn, 2018-04-24:
Did you also have this with the 3.3.0-BETA ?

supersoulman, 2018-04-24:
I installed 3.3 stable on macOS 10.13.4 and all user data and profiles were cleared out as well.
I reinstalled 3.2 and the data did thankfully reload correctly

photogrammer, 2018-06-27:
I'm having same problem with Mac version. I'm running 3.2 where I have carefully customized many settings, materials, etc... imagine my horror as I upgraded to 3.4 and see all my tweaks GONE. I restored the 3.2 app and all is well. The migration of settings should be automatic as it is with say, Photoshop. Yes, Cura is free, but I would happily pay for it if these little glitches were fixed! :-)

Appesteijn, 2018-06-27:
@photogrammer I can totally understand that, is it possible for you to zip your config folder and share it with us?

photogrammer, 2018-06-27:
would that be the "cura" folder in ~/Library/Application Support/ ?? Mine is 4.38GB!!! Yikes!

Appesteijn, 2018-06-27:
Wow, that is a lot of data indeed. We would mostly need the 3.2 folder that is in there.

photogrammer, 2018-06-27:
It seems the cache directory is the culprit, shall I delete all instances of that directory?

photogrammer, 2018-06-27:
I'm going to guess that the cache folder is not purged  before the previous settings are backed up.. that's what I'm seeing the older version folders. The current version cache folder is a lean 1-2MB

photogrammer, 2018-06-27:
[image]
This is a screenshot of the 3.2 folder

Appesteijn, 2018-06-27:
> It seems the cache directory is the culprit, shall I delete all instances of that directory?
Be sure to move them before deleting, but the cache should be safe to remove.

Why do you have the 2.6, 2.7 3.0 and 3.1 folders in the 3.2 folder?

photogrammer, 2018-06-27:
I have NO idea, I didn't put them there. :-) This is the first time I've rummaged around in "cura" folder.

photogrammer, 2018-06-27:
[3.2.zip](https://github.com/Ultimaker/Cura/files/2142866/3.2.zip)
Here's the 3.2 folder, I removed the old version folders.

Appesteijn, 2018-06-29:
Devs, CURA-5519

photogrammer, 2018-06-29:
I figured it out! My "cura" directory had an old copy of what was normally in the version folder, perhaps an old old version didn't use a subdirectory with version number? I removed those files so only the "3.2" dir was there, ran 3.4 and all works now! Clean out the lint is the watchword for the day. :-) Thanks for your help, when you were able to load and see my settings I figured it was something gummed up here and it was.

photogrammer, 2018-06-29:
My theory is that if Cura finds the config files "loose" in the "cura" directory it uses that and creates a dir for the new version, copying over the settings but does NOT remove these loose files/dirs. That explains why my 3.2 dir had copies of 2.6, 2.7, 3.1 settings. This is probably a rare problem now because old versions aren't being downloaded, only happens for those of us who have ever used an old version.

diegopradogesto, 2018-07-04:
The problem was that in case we have a configuration file from a version before 2.7 (all the files in the cura folder) and also a folder with an upgrade to (let's say) 3.2, when starting a new version such as 3.4, then instead of picking up the 3.2 folder to upgrade, Cura picked up all the contents in the cura folder, creating a mess with the files and make it inconsistent.

FYI, this fix was done in https://github.com/Ultimaker/Uranium/commit/e20e550fc4017b2a8c92ad5b569627fe6432bf97

photogrammer, 2018-07-04:
Thank you so much for fixing this! I will no longer worry about making custom settings for fear of losing them like I have the past year!!! :-)

diegopradogesto, 2018-07-04:
Now you can also use our new **Cura Backups** plugin, that you can find in the Toolbox. Try it out and save your data when you think is in a good point, and restore it whenever you need, also in different computers.