Ultimaker/Cura #3708: Cura 3.3.0

Kmarkopl, 2018-04-24:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Kmarkopl, 2018-04-24:
Cura 3.3.0 
Octoprint camera image not showing

ianpaschal, 2018-04-25:
Please fill out the full issue template including steps to reproduce, logs, and screenshots, or the issue ticket will be closed.

Kmarkopl, 2018-04-25:
Am not familiar with the system here, so will describe the problem best as I can.
Or maybe some one can do for me please. 

Here is the problem .
I have installed new Cura 3.3.0.
All works well, except I don't have webam picture in monitor window, after connecting to Octoprint.
The connection to Octoprint i established and all works, but no camera picture.

Have reinstalled plug-in, nothing works.

Eventually have removed Cura 3.3.0 and reinstalled Cura 3.2.1, and all works as it should be.
Won't be using Cura 3.3.0, till the problem is sorted !!

ianpaschal, 2018-04-25:
> Am not familiar with the system here, so will describe the problem best as I can. Or maybe some one can do for me please.

When you create a new issue there's a template already in the text box with a warning that if you do not use this template, we cannot debug your issue.

I suspect this is fixed in 3.3.1, being released today.

closed 2018-04-25 by ianpaschal

fieldOfView, 2018-04-25:
I would be much surprised if the 3.3.1 update fixes @Kmarkopl's issue.

Kmarkopl, 2018-04-25:
Sorry ! 
But I don't know what do you expect ! 

Cura 3.2.1 all works as it should be .
Cura 3.3.0 No Camera  Picture in monitor window !

Nothing was changed except new Version of Cura.

How that is not clear ?!

ianpaschal, 2018-04-25:
Well there is a template of what we expect when you make a new issue. 😅

Include Cura projects, logs, screenshots, and detailed steps of what you were doing.

fieldOfView, 2018-04-25:
Continued here: https://github.com/Ultimaker/Cura/issues/3725