Ultimaker/Cura #3709: Characters jumbled under Layer Height

GuindonDesigns, 2018-04-25:
[image]

Ellecross, 2018-04-25:
YIkes. @GuindonDesigns, which printer does this happen to? It looks like all the profiles are present for those.

In the meantime, as workaround, go to Custom tab and on the profiles drop-down, you can select the profile in the menu. Like so: 

[image]

ianpaschal, 2018-04-25:
@GuindonDesigns, Please fill out the issue template (all information is relevant 95% of the time). We can't spend time guessing at the issue or even what version you're using.

If you do not fill it out we'll simply have to close the ticket.

ianpaschal, 2018-04-26:
When you press new issue there's a lot of text in the issue box, which you must have deleted. It looks like this:

```
<!--
The following template is useful for filing new issues. Processing an issue will go much faster when this is filled out, and issues which do not use this template WILL BE REMOVED.

Before filing, PLEASE check if the issue already exists (either open or closed) by using the search bar on the issues page. If it does, comment there. Even if it's closed, we can reopen it based on your comment.

Also, please note the application version in the title of the issue. For example: "[3.2.1] Cannot connect to 3rd-party printer". Please do not write thigns like "Request:" or "[BUG]" in the title; this is what labels are for.

It is also helpful to attach a project (.3mf or .curaproject) file and Cura log file so we can debug issues quicker.
Information about how to find the log file can be found at https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations. To upload a project, try changing the extension to e.g. .curaproject.3mf.zip so that github accepts uploading the file. Otherwise we recommend http://wetransfer.com, but other file hosts like Google Drive or Dropbox work well too.

Thank you for using Cura!
-->

**Application Version**
<!-- The version of the application this issue occurs with -->

**Platform**
<!-- Information about the platform the issue occurs on -->

**Qt**
<!-- The version of Qt used (not necessary if you're using the version from Ultimaker's website) -->

**PyQt**
<!-- The version of PyQt used (not necessary if you're using the version from Ultimaker's website) -->

**Display Driver**
<!--  Video driver name and version -->

**Printer**
<!-- Which printer was selected in Cura. Please attach project file as .curaproject.3mf.zip -->

**Steps to Reproduce**
<!-- Add the steps needed that lead up to the issue (replace this text) -->

**Actual Results**
<!-- What happens after the above steps have been followed (replace this text) -->

**Expected results**
<!-- What should happen after the above steps have been followed (replace this text) -->

**Additional Information**
<!-- Extra information relevant to the issue, like screenshots (replace this text) -->
```

Obviously most of this information was missing from your initial post, some of which you've added, but we still need a `.curaproject.zip`, steps of how to recreate the problem besides the screen shot of the result, and the Cura logs.

GuindonDesigns, 2018-04-26:
Nvm...you have a new version to download, I'll report on that one, you
obviously are a bit too high on the "Tech Horse" to actually take what
I've given you seriously.

All I did was open the damn software, no projects, I didn't even configure
my printer, it was there already.

End of discussion.

Cheers, have a great day.

ianpaschal, 2018-04-26:
> You obviously are a bit too high on the "Tech Horse" to actually take what I've given you seriously.

I'm sorry you feel that way. But we are pretty explicit about what what we need in order to our job and you refuse to give it. We appreciate you taking the time to file a bug report but if it doesn't contain the information we need and you (still?) are not filling in those blanks, we can't do anything.

Do you think we're just going to try randomly pressing buttons and modifying configurations until we randomly happen to get your results? Of course not. That's why we need information like debug logs.

> End of discussion.

I'm sorry to hear that. I'd like to know what's causing this but we can't begin to figure that out without log files.

closed 2018-04-26 by ianpaschal

Ghostkeeper, 2018-05-25:
It looks like it somehow got additional profiles for a different extruder or a different material in that slider.