Ultimaker/Cura #3710: [3.3] Cura thinks model is to large and won't slice

MaxKielland, 2018-04-25:
**Application version**
Cura 3.3

**OS Version**
Windows 7, 64bit

**Graphic Driver**
NVIDIA 23.21.13.9135

**Printer**
Ultimaker 3
[UM3_bobin_v2.curaproject.3mf.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1945079/UM3_bobin_v2.curaproject.3mf.curaproject.3mf.zip)

**Steps to Reproduce**
Just load the model.
Center it
Move it to X,Y,Z [0,0,0]
Model size reported by Cura: 211.0 x 211.0 x 15.0 mm
Plate size set in Ultimaker 3 setting: 233 x 215 x 200 mm (default settings by Cura)

**Actual Results**
It complaints that it will not fit even if the dimensions are smaller than the maximum defined for Ultimaker 3.

**Expected results**
It would have not complained and sliced the model.

**Additional Information**
If I resize the model to 90.47% (190.9 x 190.9 x 13.6) it fits and slices, but not at 90.48% (190.9 x 190.9 x 13.6) and above. No Build Plate Ashesion has been used to minimize the size.
Log File: [cura.log.zip](https://github.com/Ultimaker/Cura/files/1945113/cura.log.zip)

Ellecross, 2018-04-25:
Hi @MaxKielland Thanks for reporting this and attaching the project file immediately. I'll check it out and get back to you!

ianpaschal, 2018-04-25:
This is definitely caused by the small disallowed area that exists even with no brim but why extruder 2 has it and extruder 1 does not I'm not sure. We'll look into it further.

Ellecross, 2018-04-25:
@MaxKielland I can slice your model without changing the dimensions. 

Can you try this please? 
1. In extruder 1 > Change the Travel Avoid Distance to 0. 
2. Change your model X position to -8. 

[image]

Model slices now for me.

diegopradogesto, 2018-04-25:
Yes, the thing here is that the disallowed area in the border is created due to the Travel Avoid Distance that is equals to 3mm in the UM3. You can tweak this setting to make your models fit in the build area. Setting the value to 0 will make your combing moves to go through the walls. Probably you don't want this, but maybe you can set the value to 1mm to keep the nozzle away from the walls.

closed 2018-04-25 by diegopradogesto

MaxKielland, 2018-04-25:
Hi, thank you for looking into this.
I managed to change the avoid distance to maximum 0.9mm and then it was sliced.

So this avoidance distance means that I can't really use the full area?

Ghostkeeper, 2018-05-11:
Yes, if you have "Avoid Printed Parts When Travelling" enabled, then it needs to be certain that it can travel around the object on both sides.