Ultimaker/Cura #3711: [3.3] Newer Marlin versions don't change bed origin on G29

tenten8401, 2018-04-25:
**Application Version**
3.3

**Platform**
Manjaro Linux KDE

**Printer**
Printrbot Simple Metal

**Steps to Reproduce**
- Update Marlin to latest build on Printrbot
- Print anything

**Actual Results**
Print gets mashed against side of buildplate

**Expected results**
Print should be centered

**Additional Information**
Here's a short video of the issue: https://www.youtube.com/watch?v=WXLWZ5NAMmo

Here's the line I commented out in the printer Gcode to fix the uncentered print:
[image]

**I will do some testing later with an unmodified Printrbot Simple to see if this line is even required and update the issue with my findings.**

tenten8401, 2018-04-25:
Alright so it appears that line isn't even needed on a unmodified Printrbot running the Marlin build that came with it...
[image]
[image]
[image]

Ghostkeeper, 2018-05-11:
Thanks for testing. I've removed the line from the default starting g-code so it'll be fixed in Cura's next release.

commit 16c9c4a3 (2018-05-11): Don't correct bed origin any more

It's not needed with current versions of the Printrbot Simple version of Marlin, or even the version that the printers are shipped with. Thanks to tenten8401 for the debugging.

Fixes #3711.
closed 2018-05-11 by Ghostkeeper