Ultimaker/Cura #3713: Show setting changes

BagelOrb, 2018-04-25:
It would be really nice to have an option to show all settings I've changed.

It could be an option in the drop down menu next to the setting search bar.

It would then make all and only those settings visible which have a [reset to default] icon next to it.

smartavionics, 2018-04-25:
Yes, this would be great.

ianpaschal, 2018-08-01:
Devs, see CURA-5355

Ghostkeeper, 2018-08-21:
Our project manager took this off of our planning. We won't get around to implementing it soon.

Ghostkeeper, 2018-12-14:
A couple of UX things to think about:
* If you don't have any changes, what will it display?
* If the value for a profile setting is currently at 0.4mm and you want to type "0.45mm", should the setting briefly disappear after displaying "0." and before displaying "0.45"?
* If you click on one of the reset buttons, should the setting disappear too?

BagelOrb, 2018-12-14:
1. the settings you already have visible
2. The setting visibility option I proposed is a one shot thing. It isn't keeping all and only hanged settings visible. It is only making all changed settings visible at one point in time.
3. No

MariMakes, 2023-05-10:
Hey,

Quick update. 

In the latest version of Cura you can see an overview of the settings by clicking the Save Icon for a new profile. You can close the window without saving. 
[image]

Personally I also really like this plug-in by @fieldOfView, that has a tab with changed settings. You can get it here: https://github.com/fieldOfView/Cura-TabbedSettingsPlugin
[image]