Ultimaker/Cura #3715: Possibility to take model preview - snapshot

owlet123, 2018-04-25:
Hi guys!

I took a look that you are working on [snapshot](https://github.com/Ultimaker/Cura/blob/master/cura/Snapshot.py) feature, so I would like to ask you how is it going? When do you want to release it? Does it have some limitations? 

I really look forward to this feature and if I can help you, let me know what I can do.

Thank you.

fieldOfView, 2018-04-25:
Snapshots are in active use now, in the UFP file format. As far as I know, UFP is a subset of 3mf, with gcode and a snapshot embedded. The snapshot is shown on the UM S5 touchscreen and in Cura Connect (UM3, UM S5).

closed 2018-04-25 by ChrisTerBeke

Ghostkeeper, 2018-05-11:
We have no plans to make an actual screenshot feature. Your operating system will usually provide that for you. The one we were working on creates a thumbnail for display on printers that support the UFP file format.

> As far as I know, UFP is a subset of 3mf, with gcode and a snapshot embedded.

The intention was to get as close to 3MF as possible, but it's not actually a subset of 3MF since 3MF requires a model file to be present and we don't put the model in the UFP files. Only g-code and the thumbnail.