Ultimaker/Cura #3716: solid view looks great, prints messed up, layer view shows messed up

superalpha0, 2018-04-25:
The solid view looks fine, print it and there are slots missing on top of model.  Layer view shows missing slots.

**Application Version**
3.3.0 and 3.2

**Platform**
Windows 10

**Display Driver**
Intel HD

**Printer**
TEVO Tornado

**Steps to Reproduce**
open solid view and compare to layer view

**Actual Results**
top features/slots are missing and filled in

**Expected results**
slots should be there, just like model view

[s.zip](https://github.com/Ultimaker/Cura/files/1947261/s.zip)

smartavionics, 2018-04-25:
You mean like this:

[image]

smartavionics, 2018-04-25:
Sorry, try again... I think the model has some problems, the solid model has some missing surfaces:

[image]

This is what freecad thinks:

[image]

fieldOfView, 2018-04-25:
The model is not "watertight" or "manifold". Or in other words: it is broken. Just because Cura is able to display it does not mean it can slice it.

ianpaschal, 2018-04-25:
You'll have to use something like MeshLab to repair it. This is not an issue with Cura.

closed 2018-04-25 by ianpaschal

superalpha0, 2018-04-25:
I see what you mean.  Thanks.