Ultimaker/Cura #3717: also cura 3.30 crashing on startup on win7 x64

codern00b, 2018-04-25:
I have upgraded  from 3.2.1, and also installed alongside 3.2.1, with same issue, it crashes on startup.
I get a crash dialogue box while the startup sequence is happening(the little dark grey window).
This is what is in the crash report: trace: Traceback (most recent call last):
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\PluginsModel.py", line 44, in __init__
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\PluginsModel.py", line 83, in _update
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\PluginRegistry.py", line 322, in isBundledPlugin
  File "ntpath.py", line 632, in commonpath
ValueError: Paths don't have the same drive
---------
 and logs:
Thread 0x00003c28 (most recent call first):
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 155 in _storeStderrToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003ea8 (most recent call first):
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 147 in _storeOutputToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000433c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001174 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003e04 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004280 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002238 (most recent call first):
  File "E:\Program Files\Ultimaker Cura 3.3\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 73 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000638 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000413c (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003e44 (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000041fc (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "E:\Program Files\Ultimaker Cura 3.3\plugins\UM3NetworkPrinting\UM3OutputDevicePlugin.py", line 274 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004120 (most recent call first):
  File "E:\Program Files\Ultimaker Cura 3.3\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00004334 (most recent call first):
  File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 364 in _logInfoWidget
  File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 214 in _createDialog
  File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 82 in __init__
  File "X:\3.3\build\inst\bin\cura_app.py", line 101 in exceptHook
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 246 in initializeEngine
  File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 765 in runWithGUI
  File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 687 in run
  File "X:\3.3\build\inst\bin\cura_app.py", line 141 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

codern00b, 2018-04-25:
3.2.1 works perfectly. I have already removed the octoprint plugin from 3.3 but still the crashing as described above.

LipuFei, 2018-04-25:
Hi @codern00b . This is a duplicate of issue https://github.com/Ultimaker/Cura/issues/3702 which has been solved. I will close this ticket.