Ultimaker/Cura #3722: [Request] Direct Download Links for logged-in users?

Snowflake6, 2018-04-25:
**Application Version**
All

**Platform**
All

**Steps to Reproduce**
Try to download a copy of Cura.

**Actual Results**
Note the same form comes up every. damn. time. and you have to fill in the same. damn. info.

**Expected results**
Site should recognize a returning user and offer up a direct download link  without a survey.

filipgoc, 2018-04-26:
In the meantime, you can use the 'I don't want to share any information' or something option to avoid filling in all the info...

Snowflake6, 2018-04-26:
@filipgoc That is *a* solution, but I was hoping for one that allowed Ultimaker to still know that i'm a customer.  I don't mind them knowing that i'm downloading the latest versions... I just want to do it without typing it in every time.

filipgoc, 2018-04-26:
@Snowflake6 I understand and agree. Hence the "in the meantime" in the original post, just in case you or someone else missed the no info option. Cheers.

closed 2018-05-24 by Ghostkeeper