Ultimaker/Cura #3724: [3.3.1] Plugin browser font color and pop after plugin update

nicolinux, 2018-04-25:
**Application Version**
3.3.1

**Platform**
macOS 10.12.6

**Printer**
Ultimaker 2+

**Steps to Reproduce**
1. Menu Plugins -> Browse Plugins.

2. Update a plugin.

**Actual Results**
The info at the bottom of the plugin browser is displayed in a black font over a dark background - not readable.

[image]

After updating a plugin (OctoPrint), the popup with the info to restart Cura is behind the plugin browser and not visible.

[image]

**Expected results**
Text should be readable and the popup should be visible in front of the plugin browser.

fieldOfView, 2018-04-25:
Known issue, will be fixed in Cura 3.4

ianpaschal, 2018-04-26:
Yup. Sorry about that. Will be fixed in the next version.

closed 2018-04-26 by ianpaschal