Ultimaker/Cura #3725: No Camera picture in Monitor window

Kmarkopl, 2018-04-25:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

fieldOfView, 2018-04-25:
Can you upload the cura.log file from your configuration folder somewhere?

Kmarkopl, 2018-04-25:
Here please https://ufile.io/k5451

fieldOfView, 2018-04-25:
Thanks. Is ```http://192.168.1.100:8080/video``` the correct url for the camera image in your network?

I think the problem is the same as https://github.com/Ultimaker/Cura/issues/3457 and an issue of Cura as opposed to the OctoPrintPlugin, but I'll try to investigate. 

For the record, the camera image works fine for me, at least until I switch to a second OctoPrint instance.

Kmarkopl, 2018-04-26:
The IP of the camera is correct.
All works on older Cura 3.2.1 . After installing latest version of Cura 3.3.0 or 3.3.1,  no camera picture.

adzule, 2018-04-26:
`2018-04-24 11:11:36,094 - DEBUG - [MainThread] UM.Controller.setActiveStage [159]: Setting active stage to MonitorStage
2018-04-24 11:11:36,160 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [277]: file:///C:/Program Files/Ultimaker Cura 3.3/resources/qml/PrintMonitor.qml:27:5: QML OutputDeviceHeader: Binding loop detected for property "implicitHeight"
2018-04-24 11:11:40,094 - DEBUG - [MainThread] AutoSave.AutoSave._onTimeout [71]: Autosaving preferences, instances and profiles
2018-04-24 11:11:45,116 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [277]: file:///%USERPROFILE%/AppData/Roaming/cura/3.3/plugins/OctoPrintPlugin/OctoPrintPlugin/DiscoverOctoPrintAction.qml:236: TypeError: Cannot read property 'baseURL' of undefined
2018-04-24 11:11:51,815 - DEBUG - [MainThread] AutoSave.AutoSave._onTimeout [71]: Autosaving preferences, instances and profiles
2018-04-24 11:12:08,295 - DEBUG - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction.testApiKey [141]: Trying to access OctoPrint instance at http://192.168.1.167:80/ with the provided API key.
2018-04-24 11:12:09,601 - DEBUG - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction._onRequestFinished [291]: API key accepted by OctoPrint.
2018-04-24 11:12:19,989 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [277]: file:///C:/Program Files/Ultimaker Cura 3.3/resources/qml/PrinterOutput/OutputDeviceHeader.qml:31: TypeError: Cannot read property 'name' of null
2018-04-24 11:12:19,992 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [277]: file:///C:/Program Files/Ultimaker Cura 3.3/resources/qml/SidebarHeader.qml:20: TypeError: Cannot read property 'length' of undefined
2018-04-24 11:12:19,994 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [277]: file:///C:/Program Files/Ultimaker Cura 3.3/resources/qml/Preferences/MachinesPage.qml:162: TypeError: Cannot read property 'connectionText' of null
2018-04-24 11:12:20,009 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice.connect [316]: Connection with instance octopi._octoprint._tcp.local with url http://192.168.1.167:80/ started
2018-04-24 11:12:20,037 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice.connect [316]: Connection with instance octopi._octoprint._tcp.local with url http://192.168.1.167:80/ started
2018-04-24 11:12:20,072 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [277]: file:///%USERPROFILE%/AppData/Roaming/cura/3.3/plugins/OctoPrintPlugin/OctoPrintPlugin/MonitorItem.qml:7:9: QML Image: Failed to get image from provider: image://camera/2
2018-04-24 11:12:20,090 - INFO - [MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device octopi._octoprint._tcp.local already added
2018-04-24 11:12:20,092 - INFO - [MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device octopi._octoprint._tcp.local already added
2018-04-24 11:12:20,265 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [681]: Set OctoPrint camera url to http://192.168.1.167:80/webcam/?action=stream`

AirmakZ, 2018-04-28:
Exactly same issue here upgrading from 3.2.1  to 3.3.1.
Stream url is OK but error "Failed to get image from provider" appears in the log.

fieldOfView, 2018-04-28:
I still have not been able to reproduce this exact issue, but do get a similar issue when there are two printers each with a different OctoPrint instance; in that case I can only get one camera to show (the one that Cura was started with as the active printer). I'll look in to fixing that, hoping that it also fixes your issue. Note that this seems to be a Cura issue, so it may not be possible to fix it before Cura 3.4.

Something you might try is switching to/from another printer while in the monitor.

peacefulhill, 2018-04-28:
I have the same issue after upgrading from 3.2.1 to 3.3.1. I only have one printer so I can't switch between different printers.

fieldOfView, 2018-05-02:
For my reference, here's a similar issue with the UM3 camera feed:
https://community.ultimaker.com/topic/22750-camera-on-ultimaker-3-not-working-in-cura-331/

Gasman2014, 2018-05-02:
Same here - macOS. Camera was working with Cura 3.2.1, now no longer works with Cura 3.3.1. Camera feed is viewable correctly at usual address and also viewable in Octoprint.

ur7x, 2018-05-03:
Same problem for me.  The camera is fine in browser and worked on Cura 3.2.1. now it's not there.

VMaxx, 2018-05-08:
Something I've noticed with this issue is that when you click in and out of monitor the error about image changes.  First pass its complaining about image://camera/2 and  image://camera/4, next click out and back to monitor it is image://camera/6 and image://camera/8 the number increases as you click in and out.

Ghostkeeper, 2018-05-11:
ShinerG on Ultimaker's forums says he found a cause and a solution/workaround: https://community.ultimaker.com/topic/22750-camera-on-ultimaker-3-not-working-in-cura-331/?tab=comments#comment-209181

Maybe that helps some people with this when it occurs in Octoprint. We have seen reports that this also happens with connections to the Ultimaker 3 camera feed but haven't been able to reproduce that ourselves yet.

ur7x, 2018-05-15:
So I have uninstalled 3.3.1 and went back to 3.2.1 and the Camera feed came back but the profiles are not backward compatible

HappyDino, 2018-05-16:
I have the same issue.
The stream url is also ok.

rneurink, 2018-06-04:
Also having this problem the url is correct but the camera wont show up.
I've had the camera show up once

`2018-06-04 11:38:54,634 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [277]: file:///C:/Users/user/AppData/Roaming/cura/3.3/plugins/OctoPrintPlugin/OctoPrintPlugin/MonitorItem.qml:7:9: QML Image: Failed to get image from provider: image://camera/26`

`2018-06-04 11:38:54,659 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [277]: file:///C:/Program Files/Ultimaker Cura 3.3/resources/qml/PrintMonitor.qml:27:5: QML OutputDeviceHeader: Binding loop detected for property "implicitHeight"`

`2018-06-04 11:38:54,703 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [277]: file:///C:/Users/user/AppData/Roaming/cura/3.3/plugins/OctoPrintPlugin/OctoPrintPlugin/MonitorItem.qml:7:9: QML Image: Failed to get image from provider: image://camera/28`

nean-and-i, 2018-06-04:
same here on macos.
+ 3.3.1 also have issues with fonts in some dialogs like plugins.
+ 3.3.1 also generates jobname prefix (CFDMP) out of type and not from printer name as previous version did 
had to downgrade to 3.2.2 which was much more stable on my end.
need to ask if you guys require some more additional help for testing before releasing?

fieldOfView, 2018-06-04:
@nean-and-i adding more problems to this issue will not get them fixed, and will only confuse things.

All, I am just a single 3rd party contributor. As I have indicated, I cannot reproduce this issue, so I cannot even begin to fix it. Thanks for the log snippets, but unfortunately they don't help me. I see the same messages on my working installation.

AirmakZ, 2018-06-04:
@fieldOfView, is there a way to release a test version including extra debug traces or similar?

nean-and-i, 2018-06-05:
@fieldOfView , sorry, that was not suppose to go into your direction, I appreciate your work!

rneurink, 2018-06-06:
@fieldOfView is there any information you need from us? it is obviously a problem with the QImageprovider so maybe the problem lies in the source code. Is there a way to get more debug information from the provider or check if the image is available. I can set up a dev enviroment if needed.

fieldOfView, 2018-06-06:
I have finally managed to reliably make it *not* work for me, looking to see if I can see why.

RudiKlein, 2018-06-10:
Same issue here. Went from 2.7 (camera works fine) to 3.3.1 (camera fail).
Would a Wireshark trace help to see what the actual string is that is sent over the network?
As a workaround for not being able to debug Cura?

fieldOfView, 2018-06-11:
I think I may have fixed the issue. It looks like the camera image did or did not show depending on how quickly Cura and the OctoPrintPlugin started (hence my trouble reproducing it). 

Please try this fix and see if it also fixes things for you:
* Go to Help -> Show configuration folder
* Navigate to plugins/OctoPrintPlugin/OctoPrintPlugin
* Replace the file named OctoPrintOutputDevice.py with this file:
https://raw.githubusercontent.com/fieldOfView/OctoPrintPlugin/3.4/OctoPrintOutputDevice.py
* Restart Cura
* Let me know if it is fixed

msticninja, 2018-06-11:
Works for me! TY!

rneurink, 2018-06-11:
That fixed it. Great work

adzule, 2018-06-11:
Also worked for me, thank you so much!

fieldOfView, 2018-06-11:
Ok, 3 confirmed fixes, I am submitting a fixed version to the plugins repository and am calling this fixed. The issue can be closed.

RudiKlein, 2018-06-12:
You nailed it! Works like a charm now.

Thanks!

HappyDino, 2018-06-28:
Works for me.

Thanks.

ur7x, 2018-06-29:
Broken again in 3.4 LOL...

fieldOfView, 2018-06-30:
@Ur7x, glad you find it funny. Technically it is not broken *again*, but it is broken *still*; the toolbox is still serving up a non-fixed version, instead of one including the fix above.

LilBub, 2018-06-30:
I agree with FoV.  It’s very frustrating that this has made it into the current release.  And when I pointed it out to Cura they quickly closed the ticket without a fix.   I’m not impressed.

ur7x, 2018-06-30:
Sorry my sarcasm was lost in my post... of course it is not funny... it is crazy... and sloppy.

voglster, 2018-07-01:
I will also confirm the fixed version of OctoprintOutputDevice.py indeed works.  I got here after searching the issue as well.

diegopradogesto, 2018-07-10:
Hey @fieldofview , this is the fix that you made in the version 3.4.1 of your plugin, right? If I'm correct, this version is now in the Toolbox, so... should we close this issue?

fieldOfView, 2018-07-10:
Yes, this can be closed

diegopradogesto, 2018-07-10:
Ok

closed 2018-07-10 by diegopradogesto

cereyanli, 2018-09-05:
This is also working on Mac OS

paubley, 2019-06-06:
Same behaviour after upgrade from 4.0 to 4.1...

fieldOfView, 2019-06-07:
@paubley, please see https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/116 for a more up to date issue. This one has been closed long ago, because at the time the problem was fixed. Apparently some people have a similar issue with the current version.