Ultimaker/Cura #3727: [3.3.1] No printcore, buildplate or printjob information in monitor screen

Sterke-Yerke, 2018-04-25:
**Application Version**

[3.3.1] and [3.3.0]

**Platform**

Windows 10
**Qt**

**PyQt**

**Display Driver**

Intel HD Graphics 4400
**Printer**

Ultimaker 3
**Steps to Reproduce**

Click on my connected printer in the monitor tab. The camera view of the printer shows up correctly, but no additional information about printer or printjob is shown on the right hand side of the screen.
**Actual Results**

It just does not show the information.
**Expected results**

It should show the information as it was shown in previous versions. I checked it wit version 3.2.1 and in this version it still works.
**Additional Information**

[image]
(Camera picture is dark, because frame light is off)

jackha, 2018-05-09:
Hi @Sterke-Yerke ,
Do you have the newest firmware on your printer? I'm not quite sure if there have been (breaking) changes in the API, but that can explain. I know that some parts have been changed and may not include all the old functionalities yet, but we're working on that.

jackha, 2018-05-14:
Ahhh we rewrote the network plugin and that was introduced first in version 3.3. This print info is however not yet implemented in the new structure. Sorry that I couldn't give you this answer right away.

fieldOfView, 2018-05-14:
Shouldn't this issue remain open until that functionality has been reimplemented then?

jackha, 2018-05-14:
Currently it's not planned and the new implementation is possibly very different (moving more to the firmware).

Sterke-Yerke, 2018-05-14:
Hi Jack,

I don’t understand why the topic is closed. The loss of the functionality in the new Cura 3.3. version is very unexpected and not documented (at least I could not find it). It took me a while to find out this issue isn’t an issue, but simply is not implemented anymore. I guess others will encounter the same.
Isn't there a posibillity to notice users this functionality is no longer available or reimplement the old structure?

I use the functionality mainly when starting up a new print to monitor the temperatures and have a quick insight in the progress. As my printer is relative far away from my working place, this works very well for me. Although I like the 3.3 version, I will switch back to 3.2, until the functonality is back. Do you have any idea when this will happen?

Greetings,
Ype Brada

jackha, 2018-05-15:
Hi @Sterke-Yerke . Good to know that you use it in this way. I've re-opened the issue again, so others can find this issue more easily. Unfortunately I don't know when it's coming back and in what form; the new implementation may be quite different. 

@LukeChen-Ultimaker just tagging you ^^

Sterke-Yerke, 2018-05-15:
@all,

Thanks for reopening the issue and the very usefull additional information. It's all clear to me now.

Greetings,
Ype Brada