Ultimaker/Cura #3728: [3.3.1] Material profiles missing

Sterke-Yerke, 2018-04-25:
**Application Version**

[3.3.1] and [3.3.0]
**Platform**

Windows 10
**Qt**

**PyQt**

**Display Driver**

intel HD Graphics 4400
**Printer**

Ultimaker 3
**Steps to Reproduce**

In a previous version of Cura (3.0) I made for instance a new material (PETG) profile from a secunteer supplier. In version 3.3.1 I want to select the material for my new print.
**Actual Results**

The material just vanished. In version 3.3.1. this material can not be found anymore. It can not be selected in the material selection tab (because it's not there) nor is the material visible anymore in the "manage materials" preference menu. The strange thing is that some other custom made material profiles still can be selected and are visible, while others dissapear. Moreover one of the materials (PLA rood from 123-3D.nl) shows up twice in the material menu.
**Expected results**

I expect that when I upgrade to the next version, all material profiles are ported. I checked with the older version 3.2.1. and here the material profile is still available
**Additional Information**

[image]
The above screen is of material profiles in Cura 3.3.1. No PETG profile of Real(123-3D.nl) visible, or the PLA material from Ice. Notice the PLA rood material from 123-3D.nl shows up twice.
[image]
This is the screen of material profiles in Cura 3.2.1.

Sterke-Yerke, 2018-04-28:
After a few days of experimenting by uninstalling and installing Cura 3.3.1 and resetting my printer to solve problem #3727, suddenly some custom materials appeared again and a duplicate material disappeard. It lools like a similar issue as #3744.
However not all materials appeard. The PETG profile is still not in the list, while present as a file in my AppData folder.

jackha, 2018-05-09:
Hi @Sterke-Yerke , Thanks for your detailed report! I've investigated this and this is the result of a bug that was recently introduced. 

Workaround (for now): 
- Go to your cura install dir, then resources/definitions. On my computer it's `C:\Program Files\Ultimaker Cura 3.3\resources\definitions`
- Open ultimaker.def.json with a text editor (make sure you have permissions to write)
- Remove "generic_petg" from the exclude list, so the line looks like this: `"exclude_materials": [ "generic_hips" ]`
- Save and (re)start Cura.

Devs: See workaround, it's a problem with the excluded_materials. It somehow also excludes this specific material which is not generic_petg.

Sterke-Yerke, 2018-05-09:
Hi Jack,

Thanks for your reply. I’m happy it’s a known issue. I can live with the work around.

Greetings,

Ype Brada

From: jack 
Sent: Wednesday, May 9, 2018 9:31 AM
To: Ultimaker/Cura 
Cc: Sterke-Yerke ; Mention 
Subject: Re: [Ultimaker/Cura] [3.3.1] Material profiles missing (#3728)

Hi @Sterke-Yerke , Thanks for your detailed report! I've investigated this and this is the result of a bug that was recently introduced.

Workaround (for now):

  a.. Go to your cura install dir, then resources/definitions. On my computer it's C:\Program Files\Ultimaker Cura 3.3\resources\definitions 
  b.. Open ultimaker.def.json with a text editor (make sure you have permissions to write) 
  c.. Remove "generic_petg" from the exclude list, so the line looks like this: "exclude_materials": [ "generic_hips" ] 
  d.. Save and (re)start Cura. 
Devs: See workaround, it's a problem with the excluded_materials. It somehow also excludes this specific material which is not generic_petg.

Ghostkeeper, 2018-06-11:
This was fixed in 3.4 (according to our issue tracker).

closed 2018-06-11 by Ghostkeeper