Ultimaker/Cura #3729: Compressed gcode file?

jossfb, 2018-04-25:
[cura.zip](https://github.com/Ultimaker/Cura/files/1949115/cura.zip)

**Application Version**
3.3.1
**Platform**
Windows 10

**Display Driver**
Nvidia - 23.21.13.9065

**Printer**
Ultimaker 3

**Steps to Reproduce**
Using Slice button then Save to removable drive or file

**Actual Results**
Saves as .gcode.gz

**Expected results**
Save as .gcode

**Additional Information**
Maybe it was when i was tweaking my settings but all i was doing was adding more options to my layer height and some of the beta settings like fuzzy skin. Cant for the life of me see where to set the default slice as file type etc.. Hoping someone can shed some light.

Ellecross, 2018-04-26:
Hey @jossfb, this is the new behavior now as compressed gcode sends faster via Network.
However, you can still save it as gcode, when you Save to File, you can still select *gcode. 
[image]

There's a small bug there though, when you select *.gcode, it creates a double extension. But it saves it as gcode file, nevertheless.

jackha, 2018-05-09:
Hi @jossfb . Maybe a stupid question, but do you have the most recent firmware on your UM3? I think the behavior of saving .gcode.gz for UM3 is correct from the Cura standpoint. We try to optimize all workflows, so if it doesn't work flawlessly with gcode.gz then we'll have to fix that (in the firmware).

Ghostkeeper, 2018-05-15:
> Is there a way to change this back?

There is no way to change the default back for Ultimaker 3 and S5. It's not our intention to (and we've received some positive response from the community on ultimaker.com about it too).

You shouldn't get those warnings though. That is a bug but we can't reproduce that here. Maybe share the file?

ChrisTerBeke, 2018-05-23:
This is now intended behaviour.

Ghostkeeper, 2018-05-24:
> It lists an error regarding it not being optimized for the printer and when i choose to ignore these and start the print anyway: the printer then doesnt show a time reading of print time as it carries out the print.

->

> This is now intended behaviour.

I don't think this is intended behaviour. It shouldn't give warnings for .gcode.gz files. I think it might have some trouble reading the metadata without libCharon?

Ghostkeeper, 2018-05-24:
I guess that the next question would be: What firmware version are you using on your printer?

Ghostkeeper, 2018-07-05:
Since we haven't been able to reproduce this and we aren't receiving any indication on how it occurred, let's assume that this is no longer a problem.

closed 2018-07-05 by Ghostkeeper