Ultimaker/Cura #3730: Anet A8 Boot Loop

k3yl0kin, 2018-04-25:
**Application Version**
3.3

**Platform**
MacOS 10.13.4

**Printer**
Anet A8

**Steps to Reproduce**
Plug in Anet to USB while running Cura 3.3.x

**Actual Results**
Printer reboots continuously

**Expected results**
Single reboot as with Cura 3.2.  It is normal for the printer to reboot once when USB is plugged in.  

**Additional Information**
This happens on 3.3.0 and 3.3.1 but not on 3.2.1

**Logs**
[cura.log.zip](https://github.com/Ultimaker/Cura/files/1949142/cura.log.zip)

Ghostkeeper, 2018-05-16:
I can't test this since I have no Anet A8. We're going to need some help with what Cura can do to fix it.

Currently Cura tries to send a g-code command to any new USB devices connected to the computer. The g-code command requests the nozzle temperature and if the response from the printer is any number that could be a normal-ish temperature then Cura assumes that it's a 3D printer on the other end. This initial connection from Cura causes the Marlin firmware to reboot to USB mode.

What can we change in this protocol to prevent Anet A8 from going into a boot loop?

ChrisTerBeke, 2018-05-23:
No response after almost a month, plus it's a 3rd party printer. Closing issue.

closed 2018-05-23 by ChrisTerBeke