Ultimaker/Cura #3731: [3.3.1] Can't upload firmware

GMagician, 2018-04-26:
**Application Version**
3.3.1

**Platform**
Windows 10

**Display Driver**

**Printer**
Custom Prusa MK2S DIY

**Steps to Reproduce**
Select printer -> manage printer -> upload firmware -> "select and confirm"

**Actual Results**
Uploading window appears but no progress

**Expected results**
Upload firmware

**Additional Information**
On 3.2 version it was working

gordo3di, 2018-05-04:
Confirmed same issue with gMax 1.5+ and Marlin 1.1.8.

Doesn't recognize a blank arduino and cant upload new firmware on others. Cura 3.2 has no issues.

Ghostkeeper, 2018-06-06:
Cannot reproduce this using an Ultimaker 2. It uploads a custom .hex file fine.

The only change I can see to this part is here: https://github.com/Ultimaker/Cura/commit/fa9dc7a1df691a282a3ebacc5b5d7bc0ba817ea3

I'd expect that to give an error in the logs though if something went wrong, since that is a Python-side change. Do you have any log files of when this happened?

GMagician, 2018-06-06:
I'll check on Friday evening

gordo3di, 2018-06-06:
We will check on our gMax and get the log file.

GMagician, 2018-06-07:
This is what log reports:

> 2018-06-07 23:04:01,819 - WARNING - [MainThread] USBPrinting.USBPrinterOutputDeviceManager.getDefaultFirmwareName [131]: There is no firmware for machine prusa_i3_mk2s_clone.
2018-06-07 23:04:01,820 - WARNING - [MainThread] USBPrinting.USBPrinterOutputDeviceManager.getDefaultFirmwareName [140]: Could not find any firmware for machine prusa_i3_mk2s_clone.

jmdefino, 2018-06-15:
I have the same problem on a Windows 7 machine with a Monoprice Maker Select plus. I have found the problem and a workaround.
When I try a firmware update the log shows:
 File "C:\Program Files\Ultimaker Cura 3.3\plugins\USBPrinting\avr_isp\intelHex.py", line 16, in readHex
    f = io.open(filename, "r")
OSError: [Errno 22] Invalid argument: '/C:/3D Printer/ADVi3pp-BLTouch-Mainboard-3.0.2.hex'

Apparently when a file is selected an extra / is added to the filename causing an invalid argument. I edited the intelHex.py file line 16 to replace the variable "filename" with "the_actual_filename.hex" and it worked fine.

fieldOfView, 2018-06-15:
Same issue with UM2+

PR #3999 (2018-08-10): Fix firmware upload on Windows
This PR fixes firmware uploading on Windows by correctly creating a local path from a url-encoded path. 

On Windows, local file URLs start with file:/// (three forwards slashes). Instead of hardcoding a fix for Windows only, this PR uses QUrl to handle this properly.

Fixes #3731 and #3987
commit 7e905461 (2018-08-10): Fix firmware upload on Windows...

by correctly creating a local path from a url-encoded path

Fixes #3731 and #3987
closed 2018-08-10 by alekseisasin