Ultimaker/Cura #3732: [3.3.1] Doesn't start after update from 3.3.0 to 3.3.1

alwashe, 2018-04-26:
3.3.0 to 3.3.1

Windows 10

update from 3.3.0 to 3.3.1

Windows 10 freezes after installation of 3.3.1
after Restart Cura 3.3.1 doen't start
Configs: [cura.zip](https://github.com/Ultimaker/Cura/files/1950206/cura.zip)

alwashe, 2018-04-26:
after uninstalling, deleting configs and reinstalling works again

ianpaschal, 2018-04-26:
It appears some of your configs became corrupted with invalid data. I'm glad it's working now, we'll have to see how we can avoid it happening again in the future.

PreownedFIN, 2018-05-19:
I also had a similar problem. The loading screen popped up and it crashed when trying to load my own quality presets.

I didn't save the error message nor the log as the dialog said they will be sent automatically.

The problem went away after I clicked the "backup and reset configs"-button.
It didn't mention anything where the configs would be backed up. So where are they backed up to?

I upgraded from 3.2.0.

jackha, 2018-05-30:
Similar as https://github.com/Ultimaker/Cura/issues/3561
Shortly we'll have a beta. If the \00\00's are still produced and if it's one of the top problems, then we'll fix it.

closed 2018-05-30 by jackha