Ultimaker/Cura #3733: [3.3.1] USB doesn't find Anet A2 with firmware Marlin 1.1.8

Robyrob, 2018-04-26:
**Application Version**
3.3.0 Beta and 3.3.1

**Platform**
Windows 10 64 bit

**Printer**
Anet A2 with firmware Marlin 1.1.8

**Steps to Reproduce**
Connect the printer by USB cable and start Cura

**Actual Results**
Cura doesn't find any printer. It worked correctly in Cura 3.2.1

**Expected results**
Cura discover my printer is connect.

**Additional Information**
I solved on myself changing  "timeout_time = time() + timeout"    to  "timeout_time = time() + 5"
in file C:\Program Files\Ultimaker Cura 3.3\plugins\USBPrinting\AutoDetectBaudJob.py   line 57
It had that value in previus version of Cura and it worked OK.

**Suggestion**
I suggest to restore a time of 5

gCreateJoshua, 2018-04-26:
Same problem, changing the file doesnt help for me

skuznetsov, 2018-05-06:
The issue with that plugin code that self._all_baud_rates are limited to the maximum of 115200 baud, and Merlin 1.1.8 has default baud setting set to 1000000.

fieldOfView, 2018-05-06:
A default baudrate of 1000000? That does not sound right.
http://marlinfw.org/docs/configuration/configuration.html
> Baud rate
> Allowed values: 2400, 9600, 19200, 38400, 57600, 115200, 250000.

But if it works, it works, I guess.

macaron16, 2018-05-08:
I did it (add 1000000 on the 2 arrays) ... and it don't works for me !

PS : I'm on MacOS.

jackha, 2018-05-09:
I'll close the issue as we're not supporting the Anet printer printing over USB. You're free to continue discussing.

closed 2018-05-09 by jackha

LCarvalhof92, 2018-05-29:
For me the solution sugested by the OP fixed my issue with another printer that also had marlin 1.1.8 while running cura 3.3.1 on my pc with windows 10 (updated)

mattbehnken, 2018-06-22:
The USBprinting plugin was the issue:

This fixed my issue with anet a8 running marlin 1.1.8 not being detected by cura 3.3.1 ! Many thanks @Robyrob I'm on MacOS so the path was a little different HDD/applications/ultimaker cura then right click/cntrl click on cura application icon and choose 'show package contents' ~Contents/resources/plugins/usbprinting/autodetectbaudjob.py I opened with 'text edit', deleted all the baudrates that it was checking for that aren't supported by my printer, then changed the timeout to 5 seconds, file/save. Restarted Cura and back in business printing from usb. I wasn't having any trouble connecting before I upgraded my LCD, less progmem available after upgrade so maybe it couldn't respond in the timeout window of 3 seconds.

https://github.com/Ultimaker/Cura/blob/master/plugins/USBPrinting/AutoDetectBaudJob.py

`
class AutoDetectBaudJob(Job):
    def __init__(self, serial_port):
        super().__init__()
        self._serial_port = serial_port
        self._all_baud_rates = [115200]

    def run(self):
        Logger.log("d", "Auto detect baud rate started.")
        timeout = 5
        tries = 2`

macaron16, 2018-06-23:
thanks mattbehnken.
It's working now.

Ghostkeeper, 2018-08-07:
I don't understand why mattbehnken's code would make it work. The `timeout` variable is completely unused, so I've just removed it: 59bb7744368626df2b5333f7b139d8c56c47b0d9

The baud rate 115200 is also the first baud rate that it would normally try in the for loop on `_all_baud_rates`. If successful the function is returned (so the singleton list behaves the same as the original list). If unsuccessful the next baud rate is tried (so the singleton list will immediately fail the connection, and the original list will continue trying some other baud rates).