Ultimaker/Cura #3734: Corrupted profile after upgrading

modem7, 2018-04-26:
Version 3.3.1

Windows 7 x64

Wanhao i3 2.1

When upgrading, it refuses to startup with the attached error message.

Error Traceback attached.

Resetting configuration works, but realistically, re-doing the entire configuration isn't what is expected. 

[image]
[crash.txt](https://github.com/Ultimaker/Cura/files/1951887/crash.txt)

Running Octoprint as well. 

Full config folder: 
https://www.dropbox.com/s/4m5tcb5l674bycn/3.3_20180426_162019.zip?dl=0

Thoughts on resolving?

modem7, 2018-04-26:
Could be related to #3706?

Ellecross, 2018-04-30:
Hey @modem7 , thanks for attaching the 3.3 config folder. I will try to replicate your issue.... but it will be easier if you attach the 3.2 config folder as well - so that we easily see what causes the crash.

diegopradogesto, 2018-05-01:
It seems that you have a custom material called `test.xml.fdm_material` without GUID tag in it (that is mandatory). May I ask where this material profile comes from? Normally adding custom profiles from not official sources raise some issues like this since they were not created correctly.

modem7, 2018-05-01:
Ah,

I think that came from an old version as at the time, the prices weren't
calculated correctly, so I created my own material that had that

ChrisTerBeke, 2018-05-02:
Closing as it's not an issue with Cura itself (but rather with completely custom files).

ChrisTerBeke, 2018-05-02:
@modem7 you can just open the file and put any valid GUID field in it. Check the other materials for examples.

closed 2018-05-02 by ChrisTerBeke