Ultimaker/Cura #3735: Cura 3.3 on Mac doesn't attach file extension .3mf

adrianmihalko, 2018-04-26:
**Application Version**
3.3.0

**Platform**
Mac Os X

**Printer**
Ender 3

**Steps to Reproduce**
Save a project.

**Actual Results**
Saved file does not have .3mf extension

**Expected results**
Saved file does have .3mf extension, so when I click on it Cura opens it automatically. Without extension it is not possible to open project file with Cura.

**Additional Information**
Nope, that's all.

ianpaschal, 2018-04-26:
Can you post a screen shot of your save dialog? I have a suspicion of the problem but seeing that will confirm or disconfirm that suspicion.

adrianmihalko, 2018-04-26:
Here you go:

[image]

adrianmihalko, 2018-04-28:
The problem is same if I save sliced model to file locally.

johnnygizmo, 2018-05-01:
This is not just with 3mf files. If you do not at .gocde to your sliced file, it saves without one. Standard behaviour should be that if the file name does not have a matching extension to the "files of type" drop down, it should be added automatically.

Ghostkeeper, 2018-05-25:
We've changed the dialogue that caused this bug completely for the next release. In all of our tests this now seems to be fixed.

closed 2018-05-25 by Ghostkeeper