Ultimaker/Cura #3738: Cura 3.3.1 fails to start

stuartsmi, 2018-04-27:
**Application Version**
3.3.1

**Platform**
windows 10 home 64  i7 6500 12GB

**Printer**
CR10

**Steps to Reproduce**
Attempt to run Cura. Fails before interface loaded

**Actual Results**
Dialog saying cura.exe has stopped. Nothing after that

**Expected results**

**Additional Information**

[stderr.log](https://github.com/Ultimaker/Cura/files/1953405/stderr.log)

jackha, 2018-05-09:
@stuartsmi that log is really beautiful! *ahum* Well it looks very peculiar. Did you have any older version of Cura installed before? In that case Cura upgrades the old configuration, but in some cases that can fail. Is it possible to share your config folder so we can investigate? Help much appreciated! You can find it here: https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

@overcookedTOFU I prefer smoked tofu, but the nick is awesome ^^. By looking at your logs, it seems like a graphics card issue. Can you try to install the newest graphics card drivers? What brand and type is it?

@1ngwpower It seems like a permission problem. What version of Windows are you running? Make sure your user RJ can write to c:\Users\RJ\AppData\Roaming\cura. Hopefully that'll solve the problem.

jackha, 2018-05-09:
@stuartsmi I can confirm the your config produces errors. In our current master version it tries to detect malformed config files and it says this:

[image]

creality_cr10_esun pla+ 0.2 vase mode: Is it something that you created, or how did you obtain it?

stuartsmi, 2018-05-09:
@jackha I've deleted the profile mentioned from my 3.2.1 and then deleted the 3.3.1 config tree.
Restarted 3.3.1 and I get the same result.
Here is the config tree after 3.3.1 recreated it.
[3.3.zip](https://github.com/Ultimaker/Cura/files/1987878/3.3.zip)
And the stderr.log
[stderr.log](https://github.com/Ultimaker/Cura/files/1987883/stderr.log)
Its been mentioned that if the config is corrupt or cura cant handle it, that it will put up a dialog box to allow you to clean or correct the config. I never see that dialog.
Very occasionally, I see the changelog message.

closed 2018-05-10 by stuartsmi

jackha, 2018-05-14:
You're welcome. Thank you for sharing your info so we can investigate :-)

stuartsmi, 2018-05-14:
I just wish I knew why it wouldn't get far enough to tell me about the corrupt config and clean it up like I had seen described. 
I've now had a chance to test it with my current profile and its working well.

Many thanks for your assistance.

jackha, 2018-05-14:
Oh the corrupt file config message is new in the current master branch, it's not in 3.3.1 :-)